df8oe/UHSDR #934: Dummy Issue, used to easily attach images to the Wiki.

db4ple, 2017-07-02:
If you need an image in the Wiki which is not available on the internet yet, upload it to this issue and link this image.

s53dz, 2017-07-20:
I think we could include this calculated S-units table to the Wiki - S Meter section.
I don't know how to do it to show nicely in text.

73 Bojan
[S-scale.pdf](https://github.com/df8oe/UHSDR/files/1161515/S-scale.pdf)

db4ple, 2017-07-20:
| HeadingA | Heading B|
|------------------|--------------|
| A text in Col 1 | A text in Col 2|
...

s53dz, 2017-07-21:
Danilo, thanks for the hint,

I managed to insert a table. But obviously there is a different support for the format in different Wiki.
So I could not do the character micro in microVolt and a Horizontal bar (there are visible ***)  in this mode. I think the formatting is quite clumsy and of course I do not know how to use it properly.

73 Bojan

DD4WH, 2017-12-06:
upload image for wiki
[image]

DD4WH, 2018-02-04:
[image]

db4ple, 2018-03-30:
2.9.19 Tool to backup/restore configuration via USB. Requires firmware 2.9.19.
This is a temporary download location!
[uhsdr_tool.zip](https://github.com/df8oe/UHSDR/files/1863252/uhsdr_tool.zip)

db4ple, 2018-12-23:
Newer version of the uhsdr_tool.exe

[uhsdr_tool.zip](https://github.com/df8oe/UHSDR/files/2705893/uhsdr_tool.zip)

db4ple, 2018-12-30:
A primer for how we use github:
[3 (!) Repositories make up a proper github.pdf](https://github.com/df8oe/UHSDR/files/2717238/3.Repositories.make.up.a.proper.github.pdf)

m-chichikalov, 2019-01-01:
[image]
[image]
[image]
[image]

m-chichikalov, 2019-01-02:
[image]