df8oe/UHSDR #937: green led off

AngelSerra, 2017-07-06:
Is it possible that the green led is turned off when transmitting? Then it turns on when you stop transmitting, like icom yaesu....

df8oe, 2017-07-07:
That was long time on my mind :)

Done.

closed 2017-07-07 by df8oe

db4ple, 2017-07-07:
Hi Andreas,

your commit (BTW, you forget to link it to this  issue using #937 in the commit message) kind of badly interacts with the low voltage shutdown notification (which blinks the green led but returns the led to a steady green if undervoltage condition disappears while blinking). In TX state if would then lead to a steady green led.

Anyway, we should switch from direct access to the leds to a module which handles led based notifications. And only there direct switching of the leds should happen, not distributed somewhere in the code. Gets difficult otherwise once we want to "re"-use the leds for multi-purpose signalling. 

73
Danilo

df8oe, 2017-07-07:
Hi Danilo,

I cannot confirm. I have tested: blinking LED works as before. May be in case of TX switching +1sec. there may be a hopper - but it is blinking anyway. Rewgardless if you TX meanwhile or not.

How can I link to issue #937 while pushing in Linux console?

73
Andreas

db4ple, 2017-07-07:
Of course during RX it works as expected but in TX IF the undervoltage goes away during TX, the green led will remain turned on until you go to RX... I know it is a rare situation (and a not very critical one).
It just illustrates the need for "dedicated" management of such kind of resources through a signal point of control. 

Linking: just write "hash" Number of Issue ( #937) and the github web interface does the magic for you.

df8oe, 2017-07-07:
Yes - if in TX (where the current is HIGHER as in RX) the undervoltage disappears, LED will remain steady ON (until next TX or undervoltage will come back when in RX). But this situation is very, very rare and if that is the only problem I do not think it is a problem... But for accurate handling: correct, it is not 100% ox :)

Tnx for the hint of adding relationships to issues in git command line!