df8oe/UHSDR #939: DSP in Wiki

s53dz, 2017-07-07:
Hi,

Since normally I do not just change the text written by somebody else I put this remark here.
I have noticed a tiny thing on a generally great Wiki page about the DSP role in mcHF FW. To my view the QSD here is not a real one as stated there, since it is not a "double balanced" but rather a balanced as Chris says on his page.

I am giving a link to a picture (I hope it works) where I examined both versions. It can be seen that there is quite a difference in using 0 deg and 180 deg input signal in all four segments of a cycle or only in two.

This hand-written picture can also be put to some place in Wiki if needed. Or converted to a schematics.

73 Bojan

http://lea.hamradio.si/~s53dz/arhiv/x-razno/1-mcHF-70MHz/QSD-3253-comp.jpg

df8oe, 2017-07-07:
Hi Bojan,

of course you are right. I have corrected it in WIKI (you can correct such things, too without the risk of deleting anything "forever"). There are many drawings on the web which show complete double balanced implementation so I think there is no need to add photos here. Many thanks for your contribution!

73
Andreas

s53dz, 2017-07-08:
Andreas,

yes, thank you, you corrected somewhat the reading but there is another one
in the next paragraph which is tied to the following calculation. As I am
not so near to that calculation and to a possible difference in the case of
balanced QSD as opposed to the double balanced one I would assume that it
should be checked by the original writer. This was actually my original
remark.

So, you see it is not my being reluctant to to things. I just like to see
things written by the so-called open society to hold the water. For many
people this would be a minor issue, for me it is a matter of credibility.
Saying this I am referring to my strong believe that something written
should have its signature and responsibility. Which is what I more and more
miss in the cloud of "Wissen".

73 GL Bojan

DD4WH, 2017-07-08:
Hi Bojan,
thanks for your comment!
Well spotted and you are right: the mcHF has a SINGLE balance quadrature sampling detector, not a double balanced.
I have changed the text and the calculations accordingly.
BTW: all the changes have their signature in github, everything is transparent and can be followed. 
Yes, and I feel responsible for the text: all the faults I made in the text are mine. ;-)
All the best 73s,
Frank DD4WH

closed 2017-07-08 by s53dz