df8oe/UHSDR #942: RTC via I2c module

g7jiq, 2017-07-08:
I am new to this group, and have been reading though closed issues and ideas,
I am sorry for bringing this up again.
I think it would be a great idea to be able to add an RTC with minimal conections.
All tho the current mod is very good and is the way to go, it is something that early builders will have to go without, As my LCD is soldered in, and this was before sockets were suggested.
I have touch screen and all latest mods, but i am reluctent to disturb the LCD.
No doubt there will be others in the same situation.
If this is possible without to much messing about with the code, I think it would be a good addition, especiailly to those who struggle with fine soldering.

df8oe, 2017-07-08:
Hmmmm... We have closed this because we do not want to support different rtc solutions. The easiest one is supported - and that's the built-in rtc of STM. You do not need to desolder LCD. The connections all can be done on patrs side. And sorry: you have to cut some fine pitch connections. But it *IS* a project that contains fine pitch at all...

s53dz, 2017-07-08:
Hi,

I have implemented the build-in RTC modification and am very pleased with the result. I have used a 32kHz crystal from an old ARM board and it is very accurate. The matching capacitors must be carefully chosen since they influence the stability and the frequency a tiny bit.

73 Bojan

g7jiq, 2017-07-08:
Thankyou for your reply.
I have an original LCD that I fitted as supplied, I beleve its parallel and not serial, so is not compatable with the RTC mod. So this is why I would have liked to have been able to fit an IC2 RTC.
No matter, keep up the good work, much appreciated, it just would have been nice to have had an RTC.

db4ple, 2017-07-08:
Yes, a soldered parallel display is a deal break.

Integrating an I2C clock is comparetively easy, since its presence can be autodetected and there is already display code for the "normal" RTC. 

So it is not a big deal as such but a relatively good project to get started with UHSDR SW development...

Would support it with advice, but would not do it. 

73
Danilo

g7jiq, 2017-07-08:
That sounds good..
Anyone game...

closed 2017-07-11 by df8oe