df8oe/UHSDR #948: In the 11m band

AngelSerra, 2017-07-10:
The legal band in 11m are 26,965 to 27,405 including the frequency 27.405mhz, but the frequency 27.405 is passed to gen is not possible to transmit in 27.405mhz.

AngelSerra, 2017-07-10:
And if you can only in this band on the left, place the channel would be perfect, since CEPT has been standardized by the band for 40 channels worldwide, are legal and unlicensed worldwide, only with maximum Of 4w in AM / FM and 12w SSB. I know there are no filters for that band but there are no filters to reduce the strong signals from the MW band.

db4ple, 2017-07-10:
I think we discussed this in #612 with you.

In the US it is not legal to use a non-certfied device for CB operations: A CB transmitter must be certified by the FCC. Read more here: https://www.fcc.gov/wireless/bureau-divisions/mobility-division/citizens-band-cb-service (Operations).
Similar laws exist in most countries. And the development team decided NOT to support breaking laws and possibly getting sued for it. 

Thank you for your understanding.

closed 2017-07-11 by df8oe