df8oe/UHSDR #949: BPF LPF

s53dz, 2017-07-10:
Hi,

I am using the WSJT-X with mcHF (FW 2.4.0) over USB and JT65 and JT9 modulation. It is recommended to use a wide-band mode and possibly AGC OFF to peek weak signals.
So I have chosen an RX LPF 5 kHz filter in SSB mode (USB). It works just fine. I did not touch the TX filter.
For example with 5W and dipole I just worked JR6FRV with sent/received report -24/-21dB on 14.078754 MHz. It was over 9100 km QSO.

Now if I choose a narrower filter it is awkward. Please see the attached document (I hope it comes thru) with three pictures of a WSJT-X at work. I disconnected antenna for the picture to be clear. They show a flat spectrum response when using 5 kHz LPF and a filter envelope (raised noise?) when using SSB filters.
Here I have used AGC - fast. Later for confirming this measurement I tried also the recommended AGC OFF. But also there, when I decreased RF gain I came to the same pictures.

I wonder what is the catch here?
[mcHF_LPF5k.pdf](https://github.com/df8oe/UHSDR/files/1137161/mcHF_LPF5k.pdf)

73 Bojan

s53dz, 2017-07-10:
And addition:
The aim of trying narrower BW was to lower the load when there are very strong signals in 5 kHz range to be able to use AGC-OFF. I expected a flat response.

73 Bojan

df8oe, 2017-07-11:
Because of the filter edges are very sharp you do have "filter ringing" at lower bandwidths. This is due to technical limitations (not enough MCU horse power for filters which do not have such high level of ringing). Maybe that is the reason that you better should use a wider filter...

73
Andreas

s53dz, 2017-07-11:
Andreas,

Thanks for the explanation. Fair enough. Yes, you are right, I cannot even use nothing else than a WB 5 kHz RX filtering for that modulation. But the aim was fair.

Since you are talking about filter ringing, it affects than also the audio quality in SSB and in CW reception!? Digital filtering is not something easy although it is all about choosing coefficients at the end of the day. Nevertheless, perhaps the filtering code needs some revisiting and if possible some simplification to lower the filter ringing. Maybe little bit less sharpness? Just thinking.

73 Bojan

PS: I think this good performance of my mcHF HW/FW in JT9 mode would not be possible if I did not mount the UI and RF board at 90 degrees to minimize the LCD noise.

df8oe, 2017-07-11:
Hi Bojan,

we have reached edge of what is technically possible at this stage. A few weeks ago in Yahoo group somewhere noticed bad suppression of wrong sideband. The reason was too flat filter. OK - it was not ringing but it was too flat. So we must sharpen it. Now suppression is ok - but filters are ringing. You can avoid that by using filters with different algorithm - but they do need much more horse power. We want to build a complete new audio path in code in the next time. Maybe because of this we can gain more free HP and implement better filters - we will see :)

73
Andreas

df8oe, 2017-07-11:
Regarding your PCB positioning: Have you tried to pull out LCD and compare noise with and without LCD? I cannot confirm differences if LCD is fitted or not... Source of noise is at another place.

73
Andreas

s53dz, 2017-07-11:
I could have tried also without LCD in but I am also quite sure that it is not coming directly from LCD itself. I would think the really guilty ones are control lines and refresh cycle. Saying that the noise should be probably less when using parallel mode, hi. And we are back in the magic circle again. As with those filters from above.

But, separating both boards really does the work.

73 Bojan

DD4WH, 2017-07-11:
Hi Bojan, hi Andreas,

I cannot interpret the pdf, because it shows very ackward things, which are not reproducable with the filters in the UHSDR software I use. 
From the pdf I can see:
BPF 3k2(1k4): no signal passes in the centre of the filter, only at the edges
LPF 5k: no signal goes through, there is total attenuation!
LPF 2k3: signal goes thru only at the edge of the filter at 2k0-2k5

That definitely is NOT the filter response of the digital filters, that is plotted there ;-). There must be something weird in the measurement setup. 

One thing you should avoid when measuring is having no antenna and no signal. Having an open antenna input leads to all kind of internal noises coming through to your graph. In order to measure it right, you need a defined noise source of moderate power, not the -125dBm internal noise from the mcHF, which is NOT white noise at all and the noise power is too low to make reliable measurements. Take a white noise source and measure again, I think the results will be very different. Also, please always give the level in dBm, I cannot see any scale on the dBm scale, so which kind of level are we talking about here?

Filter ringing is not apparent in the graphs, I cannot see why this should  be a problem here. From my point of view, the higher response at lower bandwidths is the internal noise from the mcHF hardware, not the filter ringing! And -as far I have understood the theory- filter ringing is never avoidable if you have steep filters. So there is no alternative way to have very steep filters without any ringing, as far as I know. Would be nice to have references on such filter algorithms, if you know of any!

So: you would have to deliver some more arguments to motivate me to work on the filter code again ;-), which I think is pretty nice already. But I am always open for good measurements and will be happy to optimize the code!

All the best 73s, Frank

s53dz, 2017-07-11:
Hi Frank,

Thanks for your comment. Well, probably it would be understandable if you knew the SW from Joe Taylor WSJT-X. I am apologizing for that. I suppose Andreas knew my point.

Pictures in my PDF attachment are three. Each one shows water-fall and the spectrum of the WS (weak signal) band. I am just pointing out the spectrum curves here!. The x-axis is marked in Hz. At y axes there is a logarithmic scale of 10 per division. The point here is not an absolute power level measurement in dBm but rather to show the filter envelope!

But it is true, if I choose AGC Manual and make maximum RF gain, then the response become flat.

If you just observe the spectrum curves you can see the flat response from 5 kHz LPF.
If you compare that flat response to a 2.3 kHz LPF and BPF you can see the difference in response.
And it is nothing that is produced from the mcHF itself in this case. At least not that much. The noise you see at ends is produced just applying the narrower filter. That is all.

And to the open antenna BNC: Yes, the graphs are the same as here, just the stations signals would be mixed with it if I connected the antenna. I made it without the antenna just to present the clear view. And to minimize the mcHF self-produced noise I separated UI and RF board at 90 degrees as I said before.

I understand that the processor power here is in question, however I can see from the pictures from WSJT-X that the same happens also in normal CW/SSB using normal AGC FAST/SLOW.

I would like someone else to do the same test and confirm.

73 Bojan

s53dz, 2017-07-11:
... and also, perhaps Danilo can make a comment on what RX filtering he was using. I have seen him on the band working JT65 / JT9 modulation from F/.

73 Bojan

DD4WH, 2017-07-11:
Hi Bojan,
thanks for your comments. I think we should be careful:
the 5k LPF graph does NOT show a flat response. It shows NO signal coming thru. Could be that the filter is blocking all the signals OR the signal is not high enough. (Because you have no antenna AND AGC on, I would say that the AGC should be OFF and you need a signal to evaluate the filter response).

You are trying to evaluate the filters without having any signal that the filters could work with (besides from the internal noise). So, do not trust the graphs, until you have a reliable noise signal as an input to the filters. 

"The noise you see at ends is produced just applying the narrower filter." --> please explain to me how a digital filter can produce "noise" that is 70dB above the band noise (see your graph no. 1) without anyone of us noticing that the filter is not working properly in hundreds of mcHFs. I really do not trust the measurement in your graphs. And it will not help if someone else does the same test without having a proper signal source for the filters!

Best would be if you use a white noise source as the input and then use audio USB to put through the audio to the PC and then measure it with an audio spectrum analyzer software.

All the best 73s,
Frank DD4WH

s53dz, 2017-07-11:
And an additional information for Frank about the absolute levels:

Using JT65 modulation you can come to -11 dB under the level of noise if using an SSB filter. Just by considering the difference in used bandwidth (180 Hz) not the transmission time.
Using JT9 you do it (only B) for additional -10 dB. It shows out as the overall improvement of 2 dB.

So when you say -125 dBm of noise floor, I say -145 dBm of usable signal. Well, considering, ... maybe I am a little too optimistic here.

So yes, I expect a 2.3 kHz LPF to have the same flat response as 5 kHz LPF at all levels!

73 Bojan

DD4WH, 2017-07-11:
Hi Bojan,
thank you for explaining the JT65 and JT9 signal levels to me.

You can expect the LPF 2k3 to have the same flat response as the LPF 5k in the UHSDR software.

However, it is up to you to convince me by appropriate measurements that this is not the case. 

At the moment you have not yet convinced me for the reasons stated clearly in my messages above.

We are doing our best to constantly improve the UHSDR software and -as you know- we like to react quickly to optimize things. But issues have to be properly measured and presented in order for me to be able to react to them.

At the moment, I can see no issue with BPF LPF.

Have fun with the UHSDR software,
Frank DD4WH

s53dz, 2017-07-11:
Frank,

Yes, I agree totally: No signal no curve. 
Here I am talking about the artefact (maybe) when observing the filter response at the same levels. So, not without signals.
Perhaps it also depends of the WSJT-X algorithms. That I don't know.
I hope to decipher this to be able to use also the 2.3 kHz LPF for the purpose mentioned.

73 Bojan

s53dz, 2017-07-13:
Hi there,

Sorry for the delay. I did make it somehow today to do a quick review of this behavior on mcHF.
All this concerns the using WSJT-X spectrum display.
I did not have time to use a Gaussian Noise at the antenna input, so I left 14 MHz antenna in.
I will do it also with AWGN later.
I have reset and rechecked the setup for a maximum sensitivity in WSJT-X.
I have checked the WSJT-X Manual where it is said that different filters can be used, however to see whole band of JT64 and JT9 sub-bands a maximum 5 kHz should be used.

Findings:
1. It is AGC dependent.
2. The worst case being at AGC manual set at 15.
3. All following tests were then made using AGC Manual set at 0. So AGC-off, Maximum RF gain.
4. Response of all LPF filters above some 4.3 kHz are the same - flat. Also with AGC on.
5. There is a kind of soft-break point of notable effect at some 2.7 kHz LPF and downwards.
6. So because of what was said the lowest BW I can use for the purpose in WSJT-X is about 4.3 kHz LPF.

73 Bojan

DD4WH, 2017-07-14:
Hi Bojan,

- we need accurate measurements without using WSJT-X or anay other program to properly evaluate response of the filters
- AGC 0 is NOT AGC OFF: switch off AGC, then measure
- You are not measuring "filter response": you are measuring the noise of the mcHF plus noise in your computer plus noise of your antenna. With your setup you cannot evaluate whether a filter response is "flat" or not

73 Frank

s53dz, 2017-07-14:
Hi Frank,

First point:
I am simply showing results of using different filters in a chain **antenna-mcHF-WSJT-X**.
Again, I am not talking about the accurate filter response measurements here. I rather rely on what you said that the SSB LPF and the 5 kHz LPF have the same (flat) response. I understand you have measured that. Fine. This is precisely what I would need for my purpose (to raise S/N a bit by lowering the bandwidth from 5 k to 2k3!) . But as I pointed out in my pictures: the filter response is reflected in the spectrum shown (although somehow inverted?).

I have to stick with using WSJT-X, since here I have more zoomed spectrum display, for example 2 kHz, 3.5 kHz. And here I have the artefact that provides me from using narrower filter. I don't know for SSB or other modes. 

Second point:
About the AGC OFF, we can find this text in the Menu description:

"Standard AGC: Automatic Gain Control Mode
setting. You may select preconfigured
settings (SLOW,MED,FAST), define settings
yourself (CUSTOM) or **use MANUAL (no AGC,
use RFG to control gain**"

So I am using the only possibility to switch off the AGC in FW Menu. Putting AGC on manual and using RFG=0. I hope I am not mistaken here.

Third point:
I agree on your third bullet point. It is of course a compound of all those signals. Perhaps "a filter response" was a strong word from me for what a measured filter response would be. Sorry if I mislead you.

By the way a dBm display of my mcHF shows a noise level < -130 dBm when I use a 500 Hz BPF and 50 ohm at input. And of course proportionally 10 dB more when 5 kHz.

And, please, explain to me, what kind of **noise from the mcHF and laptop and the antenna** could have such an envelope that is so tight to the characteristics of a chosen filter - cut-off freq or center freq of BPF could easily be seen? I can not figure that out and I do not have answer to that.

73 Bojan

s53dz, 2017-07-14:
Since there is no progress on this item and the behavior even in the newest WSJT-X release 1.8.0-rc1 is the same I am staying with the 5 kHz LPF. This works just fine.

With this I thank you for the discussion and am closing this issue.

73 Bojan

closed 2017-07-14 by s53dz

df8oe, 2017-07-15:
I never have understood what is the problem here. I am working with WSJTX (self compiled versions for Linux) many times and I can use every filter bandwidth I want. Noise (or signal) floor always is flat with steep edges at filter corners - everything works as expected. All filters do have passthroughs as expected and work fine using WSJTX. So I cannot see where progress in this should / could be. All is working as expected...

73
Andreas

s53dz, 2017-07-15:
Andreas,

Perhaps, I am just guessing here, you did not noticed it. At AGC-0 it is really small.

Could you please do this exercise for me:
To really exaggerate the point, please use AGC manual and set the level to AGC=15 to 20.
Then see the spectrum on WSJT-X display. If that would be a flat or semi-flat, then I would be puzzled even more. Hi.

TNX 73 Bojan

df8oe, 2017-07-15:
Hi Bojan,

yes: operating conditions may give the reason... I never, never would work in AGC MANUAL if I am not always present at the radio to correct it! And I never use AGC MANUAL at all :) So I do not know how to switch to AGC MANUAL. I only can switch to AGC STANDARD or AGC WDSP (which I prefer). What shall I switch to?

73
Andreas

df8oe, 2017-07-15:
Bojan, things that are obsolete (but still implemented) are "out of my mind" ;) AGC Standard is OBSOLETE - I have not used it since months so it does not exist for me any longer :) I am running WDSP AGC which is lightyears better than Standard AGC - under all working conditions I have tested.

If you set AGC Mode to MANUAL AGC is OFF regardless of your AGC setting (value). There is no automate so MANUL == OFF. Now you can set your gain using encoder 2 that it gives best result for your noise floor/signal ratio. At all stages I can see the signals on WSJTX it is always flat (regardless of filter setting). But if I optimize AGC (confusing naming - RFG  or MGC (ManualGainControl) will be better here)and wait half an hour, noise floor possibly has changed so it is strongly recommended to work WSJTX using AGC ON.

Again: I cannot see any weard frequency responses in WSJTX when I do control gain manual. "0" does NOT mean AGC OFF! If you set mode to manual, then AGC is already OFF. If you turn down your setting, you will increase digitally noise of A/D converters and other unwanted signals do appear.

If you reduce gain more and more you are reaching the state where you do not have any reception at all. What you see now are undefined states. So cranc up your gain that your signals get reasonable hight. At this state S/N is defined by first input amp (BFQ) and that is the best position for doing WSJTX.

73
Andreas

s53dz, 2017-07-15:
Andreas,

Thank you again for clearing things up.
Yes, MANUAL is certainly not AUTOMATIC. This I meant when I said set AGC to MANUAL. Then of course it is no more A(GC) and then the RF gain is simply set by RFG, no matter how and where it is implemented in SDR. And by RFG=0 I meant the maximum RF gain (which of course could be too much for the circumstances).

Now, because of your last reply that the response you get is flat for all implemented filters, I knew that I am missing something here, so I made a step by step check again. 
Et voila!
I found that I have had checked the option **Flatten** from the beginning, which means that the spectrum display baseline is flattened by recalculation in the displayed interval. And as the displayed area was 5 kHz, there it was flat anyway. When unchecking the option I got the wanted and expected behavior.

About using AGC or not it depends. If you want to control the reception of a really weak signals than you have to use appropriate BW and appropriate gain. Here the WSJT-X team would agree.

Otherwise, I have checked both: the new AGC, you pointed out, and manual mode and I can say this is it. Both.

I owe an apology also to Frank to invoke him about this issue. It was only my not knowing a tiny and precise WSJT-X functionality.

TNX 73 Bojan

DD4WH, 2017-07-15:
Hi Bojan,
no need to apologize. HAM Radio is all about learning and I (we all ;-)) learned a lot about WSJT-X with this issue.
And if it had been an issue with the UHSDR software, I (and also the other programmers) would have been highly motivated to try to improve the functionality. But in order to do that we always need the most precise and concise issue description so we know where to start optimizing.
Have fun with the UHSDR software!
73 de Frank

df8oe, 2017-07-15:
Hi Bojan,

very good that this issue is tracked back to the root problem and gone now. From my view no need to apologize - it was an issue you have observed (but your logical path to isolate it splited from reality). The more we are using different SDR or digital modes software our understanding can (and will) grow. So it is important to make issue reports and it is satisfying to get a solution - whereever the reason for the issue is located :)

Have a nice weekend
73
Andreas

s53dz, 2017-07-15:
Hi,

really just to conclude this excursion in a good spirit, what is irony:

1. I searched for a SDR radio to be independent from a PC and found this excellent mcHF project. 
But you know what, now I am using JT65, JT9 and FT8 modes which all need **EXTERNAL** computing.

2. I was trouble shooting in searching a flat response in this issue.
But, you know what, I just had to **DISABLE** a Flatten option.

Nice weekend,
73 Bojan S53DZ

DD4WH, 2017-07-15:
:-)
73 de Frank

df8oe, 2017-07-16:
Hi Bojan,

I have had the dream to use JT65 on mchf without needing an external PC, too before I started building one. This dream lasts some months - until I calculated that the amount of RAM which is present can't hold digitalized audio of a complete reception. We looked for a pin compatible MCU what does have more RAM and found one (256KB instead of 192KB). But again calculation tells there are missing some KB :( Our new project OVI40 runs a MCU with >=512KB RAM - and **will support** these very interesting modes! Again my dream is up-2-date: plug in a GPS stick to USB (for getting accurate time) and doing these slow modes portable without the need of an external PC and / or internet access. It will become true...

73
Andreas

s53dz, 2017-07-17:
Hi Andreas,

This is a great idea, however, I am afraid, to do that you would need at least a second ARM built-in.
On a WSJT-X page you can find also these requirements to run the program:

> 1.5 GHz or faster CPU and 200 MB of available memory. (MSK144 especially benefits from a multi-core CPU)
> Monitor with at least 1024 x 780 resolution

73 Bojan

db4ple, 2017-07-17:
Well,
I have been running Raspberry Pi 3 (1Ghz 4 cores) with WSJT-X for about 2 weeks now and have to say, it had to struggle to decode JT9 in the available 9s when lots of QSO & CQ calls were happening. But for a few, no problem. So I guess, with a limited amount of QSO signals decoded (e.g. by restricting the "passband" to 500Hz.  AND/OR reduced decode quality (WSJT-X has 3 of them), it could be feasible. 
Also calling CQ and decoding a signal signal on a given frequency should be doable. 
I am not sure about the RAM requirements, if we would leave out the UI part from WSJT-X, i.e. how much of the 200MB are "just for the show".

But meanwhile I would recommend to think of combining RPi (with a small scale display, I use the 7" one) plus the mcHF. Makes still a small package and could be driven from a battery pack for a few hours. 5V USB batterypacks are fairly cheap. Even the more powerful variety. 

73
Danilo