df8oe/UHSDR #953: EEprom Fails

g7jiq, 2017-07-14:
I have come across a problem with saving settings. ( 241026 fitted,)
All ver up to D2.5.17 save settings ok, (text in green)
The ver D2.5.18 and D2.5.19 both fail (text in red)  
This shows on powerdown, with delay

Anyone any ideas why this is...

Cheers in advance...

db4ple, 2017-07-14:
Hi,
this happens because there is some problem while saving. 

Previously (before 2.5.18) we did not detect the problems (so no error message). Now we integrate proper detection of problems and report them. So even if this seems like a software problem, it is rather an improvement which tells you something probably did not work correctly for quite some time. The recommended solution is to reduce the I2C speed to the default settings for the I2C2 Bus (100kHz). If you already have 100khz, you should try to go down in speed. The settings can be found in the Debug/Expert settings menu. Please report if this solved the issue.

In case everything is okay, you will get the same message as before (in green). We tested this and on our machines it does work correctly.

Thank you,
Danilo

g7jiq, 2017-07-14:
Hi
Thank you for your reply and help.
I have lowered the I2C2 Bus speed to 100khz (was 400khz lol) and it solved the probem.

Thanks again

closed 2017-07-14 by g7jiq