df8oe/UHSDR #954: Strange error handling in  UiWriteSettingEEPROM_Filter

AsbjornPettersen, 2017-07-14:
Strange error handling in this function:
It will works but it's hard to read.

static uint16_t UiWriteSettingEEPROM_Filter()
{
    int idx, mem_idx;
    uint16_t retval = HAL_OK;

    for (idx = 0; retval == HAL_OK && idx < FILTER_MODE_MAX; idx++)
    {
        for (mem_idx = 0; mem_idx < FILTER_PATH_MEM_MAX; mem_idx++)
        {
            retval = retval == HAL_OK && UiWriteSettingEEPROM_UInt16(EEPROM_FILTER_PATH_MAP_BASE+idx*FILTER_PATH_MEM_MAX+mem_idx,
	      ts.filter_path_mem[idx][mem_idx],0);
        }
    }
    return retval;
}

typedef enum
{
  HAL_OK       = 0x00U,
  HAL_ERROR    = 0x01U,
  HAL_BUSY     = 0x02U,
  HAL_TIMEOUT  = 0x03U
} HAL_StatusTypeDef;

findings:
1. retval used as a boolean (false, true) -> (0,1)
2. retval can only be 0 or 1. (HAL_OK, HAL_ERROR)

Examples of possible retval's:
UiWriteSettingEEPROM_UInt16 return HAL_OK (0)-> retval = (true) && (0) = false(0) = HAL_OK
UiWriteSettingEEPROM_UInt16 return HAL_ERROR (1)-> retval = (true) && (1) = true(1) = HAL_ERROR
UiWriteSettingEEPROM_UInt16 return HAL_BUSY (2)-> retval = (true) && (2) = true(1) = HAL_ERROR
UiWriteSettingEEPROM_UInt16 return HAL_TIMEOUT (3)-> retval = (true) && (3) = true(1) = HAL_ERROR 	

Improved function could be:

static uint16_t UiWriteSettingEEPROM_Filter()
{
    int idx, mem_idx;
    uint16_t retval = HAL_OK;
    
    for (idx = 0; idx < FILTER_MODE_MAX; idx++)
    {	
        for (mem_idx = 0; mem_idx < FILTER_PATH_MEM_MAX; mem_idx++)
        {   
            retval = UiWriteSettingEEPROM_UInt16(EEPROM_FILTER_PATH_MAP_BASE+idx*FILTER_PATH_MEM_MAX+mem_idx,
	      ts.filter_path_mem[idx][mem_idx],0);
            if (retval != HAL_OK)
	      return retval;
        }
    }
    return retval;
}

db4ple, 2017-07-14:
Hello Asbjorn,
well spotted (it is a failed copy and paste which accidentially worked) but your solution is not in line with our practice:

There should be only one return in a function.
See my coming commit.

73
Danilo

AsbjornPettersen, 2017-07-14:
My "solution" is just an example. The code look OK now. Great,

df8oe, 2017-07-14:
Hi Asbjorn,

we have implemented a "Hall of Fame" where all contributors are shown. Of course your contribution was useful and appreciated so you are listed there, too ;) Do you have a callsign? If so, we can add it.

73
Andreas

AsbjornPettersen, 2017-07-14:
No , i have no callsign yet, but i'll let you know when i get one :)

df8oe, 2017-07-14:
:) very good - sounds that getting one is on your 2do list...

closed 2017-07-25 by AsbjornPettersen