df8oe/UHSDR #956: Updating the Bootloader through USB - How to?

sv1onw, 2017-07-17:
I have a question for my mcHF

I have installed DF8OE Bl 2.0.1 and have firmware 1.5.8 running.
I want to update to 2.4.0 and the higher version of Bl.
My version of UI is with 512 KB of Flash and 192 KB of RAM and the EEPROM is 24xx01/128B as I read from the System Info.
Due to sightseeing issue I got my mcHF ready built and unfortunately I am not able to make any soldering changes on SMDs or solder a newone.

I have a question about the Bl update from 2.0.1 to higher version.
I have read the instructions, but it is still not clear to me how to make the update using the bl-mchf.bin from the USB flash drive.
From my understanding I should update the bootloader first and then update the new firmware. Do I put just the bl-mchf.bin file in the root of the USB or what?

Can someone or Andreas help me with that.
Thanking in advance,

Konstantinos, SV1ONW

db4ple, 2017-07-17:
Short answer: No. Updating the bootloader using bl-mchf.bin using the USB drive method is not possible with any current or older bootloader. For bootloader 2.0.1 there are two ways: using jumper on J6 (and Band+ during start) plus bl-mchf.dfu and DfuSeDemo or with a ST-Link and a debugger connection to the UI board. 
Once updated to the newest bootloaders you can do further update the bootloader using the DFU method as described in the Wiki (bl-mchf.dfu is used in that process). 

However, you may still use only the new firmware but keep the old bootloader. There are quite a number of nice things regarding updating built into the new bootloader but once the firmware is started, the bootloader doesn't contribute to the functionality of the mcHF as such. Although I do recommend to update to a newer bootloader just for the convenience of the "cable-only" update capability.

73
Danil

df8oe, 2017-07-17:
Additionaly info:

you do have no serial EEPROM (24C01 == "NO") installed. I don not know for how long you have saved to virtual EEPROM inside STM - but it may be if you do not change 24C01 to a supported EEPROM (like 1025 or 1026 - regarding your PCB version) in the next time you have to solder 100 pins - when STM32F4 is fried because of too many write accesses ;)

So its your decision:
Solder now 8 pins or solder in ??? days/weeks/months 100 pins.

73 de
Andreas, DF8OE

sv1onw, 2017-07-17:
Thanks a lot to all for your very prompt support and recommendations.
I will follow seeking the assistance of a fellow ham with better than mine
eyes.
73
Konstantinos

sv1onw, 2017-07-18:
Thanks a lot Danil.
Everything is clear for me now.
73
Konstantinos

closed 2017-07-29 by df8oe