df8oe/UHSDR #957: Power off problems

la2fda, 2017-07-17:
Yesterday I upgraded to BL: 3.4.0 and FW: D2.5.20
FYI:
The mcHF now have random problems with switching off.
"Saving settings finished" is the last line in the display, and it is never powering off. It is not reacting on any keypress, and I have to unplug the power plug to get it off.

db4ple, 2017-07-17:
Well,
it does not react to keypresses at this point because the software basically waits for the power to disappear.
Which it does not in your case all the time (the power does not disappear). We recently made a change in that area which was supposed to improve the switching off reliability (it actually did work in my test cases better than before) but this may cause the problems for you. What you describe indicates a potential hardware problem. You may have some soldering issue (some unwanted connection between traces which are sufficiently low in resistance but not a short) etc. which in combination with other external influences may keep the power-off signal for the LM2941 from reaching the "power-off" level sometimes.
Please check the voltage level of the pin2 of the LM2941 regulator when the device is powered-off but "not powered-off".  When powered on the voltage is near 0V, when turned off, it needs to be above 0.8V - 2V (depending on the condition), so to be safe, it needs to be above 2V (typically 1.3V should be enough). See the datasheet for more information. 

73
Danilo

df8oe, 2017-07-17:
It sounds like a hardeware issue. When mcHF is powered down STM releases the output that tears pin2 of 8V regulator down so 8V is vanishing - not at your mcHF... Why?

73
Andreas

la2fda, 2017-07-17:
Thanks for quick reply. I will make some checks later on, and come back with the result. 
It is not happening very often, and is not a showstopper for using the tranceiver.
73 de LA2FDA, Stein

df8oe, 2017-08-05:
Hi Stein,

do you have made tests? Does the problem exist any longer?

la2fda, 2017-08-06:
Hi Andreas.
I have just arrived from vacation this evening and have not been able to do any measurements yet, but I just noticed the problem is still there ( Still using the same fw) . Let me come back with some measurements as earlier described.
73
Stein

DL3AAA, 2017-08-18:
I have the same problem like LA2FDA also with myself it has happened 2 times. Me 

73 
Dietrich

la2fda, 2017-08-20:
I have been using the mchf outside the box today (ready to do the measurements), and so far I have not been able to observe the failure. Will continue to watch it.
What I can tell so far is that under normal operation the voltage level on pin2 is ON=0.2v, OFF=3.0v
Input voltage is 14.1v (pin4) and output voltage is 8.4v (pin5)

la2fda, 2017-08-20:
Now I run up to the situation again:
The pin2 voltage is not rising at all when off button has been pressed and the display is showning the "final text". It is still 0.2v in the hang situation.
Taking the powerplug and starts up again makes the mchf work as normal again.

73
Stein

la2fda, 2017-08-29:
I have now done the measurements on the D2 Diode:

-------------------------------------------
Normal operation when all is working:
<table>
<tr><td> Status    </td> <td> D2(PowerButton)</td><td>  D2(PowerDown) </td></tr>
<tr><td> ON         </td><td> 1 </td><td>     0     </td></tr>
<tr><td> Press Button   </td><td>    0         </td><td>        0      </td></tr>
<tr><td>OFF          </td><td>           1          </td><td>       1      </td></tr>
</table>

Level 0 = 0.2v and Level 1 = 2,6v

<br>
Operation when it fails to power down:
<table>
<tr><td> Status    </td> <td> D2(PowerButton)</td><td>  D2(PowerDown) </td></tr>
<tr><td> ON         </td><td> 1 </td><td>     0     </td></tr>
<tr><td> Press Button   </td><td>    0         </td><td>        0      </td></tr>
<tr><td>OFF          </td><td>           1          </td><td>       0      </td></tr>
</table>

Level 0 = 0.2v and Level 1 = 2,6v

<br>
Display: Saving settings finished
Still able to toggle D2(PowerButton) between 0 and 1 by pressing button, - No other reaction.

73 
LA2FDA, Stein

df8oe, 2017-08-30:
I am nearly sure your 3.3V supply (digital path) is not falling at the neccessary speed. Normally when mcHF is powered down 8V is disbled because of voltage on D2(PowerDown) rises. This causes dropdown of 5V and of both 3.3V. If your 3.3V stays too long alive MCU tries a reset and hangs somewhere in the code because of voltage was high enough to do some stepts but too low to do "correct steps". So it is hanging at a non-reproducable place...

Can you confirm that when error occured the voltage at D2(PowerDown) was rising a short time? If you can confirm this I think there are some capacitors which impacts fast falling of 3.3V....

73
Andreas

db4ple, 2017-08-30:
As  I said in an earlier the processor reconfigures pin2 from output to input with pullup. And then enters an endless loop.
There is no way that the processor is not doing this and only one way to change this (reset). If the processor goes into reset due to falling voltage, it should not leave it, there is a hysteresis, so even if the volatge rises a little after processor goes into reset, it should not cross the hysteresis threshold.

As Andreas said, one way to find out is the measurement of the pin2 over time with an oscilloscope. The other is to use a ST-Link debugger to acquire the processor state and see if it is in a rebooted state or still in the endless loop.

la2fda, 2017-08-30:
I am using a dual channel oscilloscope for this measurements, and I will try to confirm the behavior.
The strange is that this problem started at a upgrade to BL: 3.4.0 and FW: D2.5.20.
In which version did you make the changes which was supposed to improve the switching off reliability ?

73
Stein

db4ple, 2017-08-30:
That was 2.3.6 June 11. That was way before 2.5.20, question is, what did you have before (bootloader & firmware version). That would allow us to track down the problem more easily. 
But a trace of the pin2 would be important to see if it briefly rises and then goes down again.  We are talking possibly only a few uS here.

df8oe, 2017-08-30:
As I take a look at the correspondent code it is *impossible* that pin is *not* set to tristate if "aving settings finished" is displayed and endlessloop is reached.

So the only possibility for a fault is that the supply voltage is not falling fast enough:
- when pin is set to tristate (and it is unconditionally set to tristate!!) 8V regulator switches off 8V
- now all voltages are falling - of course digital 3.3V does, too
- but if this voltage does not fall fast enough MCU may settled in a state where it does uncalculable things like resets, jumping to any flash address etc. If it does it maybe that pin again becomes LOW. And now you are stuck at this point.

Do you have bigger capacities at voltage on/off pin of 8V regulator? Or at 8V? or at 5V? or at digital 3.3V? If one of these voltages do not fall steep enough that can be the reason...

db4ple, 2017-08-30:
Hi Andreas,
the STM32F cpus have a decent reset circuit, they will not do random things as the slowly falling voltage at power cut off is a known challenge. Cutting power is only a problem when writing to flash, but we covered that. Everything else you described would mean STM failed big time in their processor design. 

When it comes out of reset (for whatever reason), it will do things but not random.  It will start from one of the boot addresses. Question is, does it go in AND out of reset?

73
Danilo

df8oe, 2017-08-30:
I think MCU *can* go to undefined state. If supply voltage is instable and has "ripples" due to non-constand current there can be misbehaviour. I have tested that as the first "reboot" issue appeared about 2 years ago. I was able to put MCU to undefined states by increasing resistance of supply voltage...

I agree it would be very helpful to identify the state where MCU is resing if issue is present. I guess that debugging connection is lost when the issue occurs. Unfortunately I do not have STLINK 2 years ago and now I cannot confirm the issue at any of my mcHFs.

Additional idea: what about "brown-out feature"? May this be related?

la2fda, 2017-09-04:
When failing to switch off.
Pressing PowerButton drops the D2-PowerButton from 3v to 0.2v
D2-PowerDown is constant on 0.2v, and does never rise to 3v.  I can not see it is moving at all from 0.2v

I am not able to remember which release I had before the upgrade, but it could possible be a versjon from before June 11th. ( I have not been using the radio much during the summer ) 

I can see DL3AAA reports the same problem.
It is not happening that often, so it is no a showstopper for me, but i am quite sure it came with a fw upgrade.

73
LA2FDA

la2fda, 2017-09-07:
[Videoclip showing scopetrace when failing](https://drive.google.com/file/d/0B8F27iz0dgsEaHN6d0JqcHZ3ckE/view?usp=sharing)
Scope ch1 (upper) : PowerButton 
Scope ch2 (lower) : PowerDown
Both channels are set up to 2v/div

df8oe, 2017-09-16:
Hello Stein,

I have made some investigations because I have got a mcHF for repair which sometimes does not power down :)

I could get rid off the problem by swapping LM2941CT. But the originally used was not faulty - so I have done some more measurement. It seems so that input current to pin2 differs due to charge of LM2941. If you got one with high current used divider R9/R10 does not work as expected and voltage on pin2 does not raise as neccessary.

I have swapped R9 to 10K and R10 to 3.3K and issue was gone using old LM2941, too. So it is a design error - not a firmware problem...

Please try and report if that solves the problem for you, too.

73
Andreas

la2fda, 2017-09-16:
Thanks for suggestion  Andreas.
I will try if I still have the problem. :-)   ( ...and I guess I have )

Last week I have started to suspect that the problem only happen if I recently had been using the radio to Tx-CW ( which I am doing most of the days. ) I have tried to stress test this,  using cw and ssb and frequently powerOff. I have not been able to have a powerOff failure when just using SSB. 

Today I upgraded the Bootloader to 3.4.1  and the FW to 2.5.101, and during the day I have not been able to trigger a powerdown-hang, - even if I have tried very hard :-)
It is to early to conclude that the problem is gone, - I will follow up with changing the resistors if the problems still here. Anyway I will report here in a couple of week or so. 
( I guess I have to change the resistors, but I want to confirm it still is failing first.

DL3AAA: Do you still have the problems, and do you have any idea of ways to trig the problem ?

73
Stein

la2fda, 2017-09-18:
Hello Andreas.
I just had a fail situasjon again, and I will now swap the two 
resistors. Will make a report when I know the result.

73
Stein

df8oe, 2017-11-01:
Some new investigations:

I have got two mcHFs which shows this behaviour. It was getting better after swapping the two resistors but it was not gone. Swapping UI board to another mcHF which does not show this behaviour leads to swapping the issue, too. So problem must be located on UI board...

I investigated the main problem is how 3.3V for MCU is gained. There are two possibilities actually outside in the wilderness:
1. sourced from 8V
2. sourced from 5V

Issue only occurs when 3.3V is gained from 8V. So if you do have a board which gains MCU voltage from 8V rail remove 4.7R resistor in front of 3.3V voltage regulator, put input of regulator to 5V on UI board (e.g. backlight Vcc pin of LCD) and swap regulator to a low drop regulator (if not already fitted). I can confirm that issue is 100% gone on all mcHFs which do have it before. And all have had gained 3.3V from 8V rail.

I cannot 100% explain but it must relate to the way voltage decreases if mcHF powers down.

73
Andreas

df8oe, 2017-11-04:
please test and report.

la2fda, 2017-11-06:
Hi.
At the moment my mcHF is dead (of other reasons), so I am not able to do 
any tests at the moment.

LA2FDA, Stein

iw5bms, 2018-01-25:
Hi Andreas, I have last mcHF v0.7 with D2.105.106 firmware and the problem mentioned above is still present. Except the 8 VDC obtained from main 12 VDC the others 5vdc is derivated from 8 vdc and 3vdc is derivated by 5vdc. I tryed to changed normal electrolitic capacitors around LM2941CT with tantalium type and add 100 nF capacitors on pin 2 of voltage regulator but without success.
Thank you, regards
Maurizio

df8oe, 2018-01-30:
solved with bootloader 3.4.6

closed 2018-01-30 by df8oe

iw5bms, 2018-01-30:
Hi Andreas and thank you for suggestion. I will upload the bootloader soon.
Maurizio iw5bms