df8oe/UHSDR #958: AGC OFF - a question

s53dz, 2017-07-19:
Hi,

Standard AGC and WDSP AGC are working very well and the latter is also settable. But, to squeeze maximum sensitivity out of the mcHF in case of weak signal reception, and prevent masking, sometimes it is necessary to use a narrower filter and a manual RF gain control. But, of course, this on HF is a very tricky thing. One reason being signal fading and another the fact that "weak signal" parts of bands are becoming crowded with high power signals ... Manual AGC mode is probably impossible to use in FT8 mode.

Now, when using the WDSP AGC there is a white marking in a blue rectangle, showing a threshold for us to see that we are on the safe side. I miss that marking in manual mode. Is it something that can be easily extended to show in manual mode, perhaps even in a different color?

73 Bojan

df8oe, 2017-07-19:
If you are in WDSP mode the blue box shows the threshold when WDSP AGC is beginning to work. In manual mode WDSP AGC never does work so the state when it would start working cannot be displayed...

So the blue box can only be present in WDSP AGC mode.

73
Andreas

s53dz, 2017-07-19:
Andreas,

> Using manual mode gives 10% better reception and 400% more problems.

I totally agree with you on this sentence. Most of the time also I am using AGC with WS reception.

But, to turn words around a little bit: A white AGC sign begins to lit when the signal overcomes the set AGC threshold (by encoder), if I have understood it correctly. The essence of my question was, if I use manual RF gain (in spite of what I said in the first paragraph) could I have an indication of a possible clipping of a signal at the ADC input. Perhaps this max. permissible level -10% could be used here as threshold.

And then again, for brainstorming purpose:
It can happen that while receiving a (very) weak signal, some in-band strong signal that is greater than threshold, could lower the RF gain (AGC activated), and it will be lower for a hold time of AGC (FAST), and thus preventing from decoding the wanted signal.
Or am I kicking the air here?

TNX 73 Bojan

s53dz, 2017-07-25:
I am closing this question since the usage of Manual RF gain is quite unique.

73 Bojan

closed 2017-07-25 by s53dz