df8oe/UHSDR #960: Back up config question

WD8BXS, 2017-07-23:
I would be neat to be able to back up and restore you configuration settings to thumb drive.

Chuck

df8oe, 2017-07-24:
Hi Chuck,

this is already on our todo list.

73
Andreas

WD8BXS, 2017-07-24:
Thanks Andreas,

Also I have one here on the bench, that has a crackling from the speaker and I see the S meter scale flashing red a lot, but not on another rig I have here.

Any ideas?

Chuck

From: DF8OE 
Sent: Monday, July 24, 2017 2:55 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] Back up config question (#960)

Hi Chuck,

this is already on our todo list.

73
Andreas

df8oe, 2017-07-24:
There can be many reasons. It is located in audio path from I2S-stream input to STM32 to rx preamp. It can be a bad soldering or a malfunctioning semiconductor (audio codec, operational amp, rx mixer) and can also be in the LO input section of rx mixer. There is no possibility to guess something out of that bunch.

Andreas

WD8BXS, 2017-07-24:
Noticing that this receiver is also showing much higher receive signals that the other one, I put the receive gain to 30 and they match there, but still getting the red scale and 
it sounds like there is a storm outside with static like crashes.

Strange that it would be so much more sensitive.
Chuck

From: DF8OE 
Sent: Monday, July 24, 2017 8:17 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] Back up config question (#960)

There can be many reasons. It is located in audio path from I2S-stream input to STM32 to rx preamp. It can be a bad soldering or a malfunctioning semiconductor (audio codec, operational amp, rx mixer) and can also be in the LO input section of rx mixer. There is no possibility to guess something out of that bunch.

Andreas

closed 2017-07-25 by WD8BXS