df8oe/UHSDR #961: No audio + error message with latest F/W

Haasoperator, 2017-07-29:
I have had this problem since every build AFTER the May 22, 2017 "daily snapshot", but with the latest F/W snapshots I am getting new messages. 

First issue: when I have installed the f/w snapshots made AFTER May 22, 2017, I get ZERO sound, and ZERO traces on the scope. Doesn't matter what band I am on.

Newest thing is, when I install the latest F/W, on the boot screen I get the following messages:

"Audiocodec WM8371 NOT detected"

"Boot delay due to error/warning" Takes several seconds to boot up, then when it is on the main display screen, I get no sound and no traces on the scope.

Attempting to reset the codec in the menu yields no results. Restarting the radio gets no results. The only thing that gets the radio functioning again is installing older firmware (prior to May 27th, 2017, which was the point at which the "newest/latest/greatest" stopped functioning in my radio)

I have lost my copy of the May 22, 2017 "daily snapshot", so now the latest working one I can find is 2.2.0 from March 25th, 2017.

df8oe, 2017-07-29:
You do have problems on I2C bus no. 2 - I am guessing due to hardware problems. But maybe you can get a workaround in firmware:

Go to configuration menu - I2C2 Bus Speed and set the value down to 25KHz. This is the slowest speed that is technical possible and was the standard speed (or should I say "standard slow"?) of older versions. If you do have proper hardware you should be able to run both busses @320KHz (I have confirmed this with all mcHFs I have seen since then, about 16 pcs.). 100KHz should be safe - and 25KHz is for "worst case hardware".

I2C1 is for SI570 and temperature sensor
I2C2 is for audio codec and serial EEPROM.

If audio codec is not detected but serial EEPROM is (both are located on the same bus!!) there can be a defect of audio codec, bad soldering around audio codec or wrong (too high) resistors @R21/R22 ui PCB.

db4ple, 2017-07-29:
Hi,
since the change of firmware creates this "problem", it could also be a firmware problem. This is relatively unlikely (in this case more users would have problems). So I assume it is the improved error detection no showing an issue with existed before probably due to hardware problems in the I2C bus. 
As Andreas wrote, if reducing the I2C speed resolves the problem, you have to check the I2C bus.

Here is a question for you: Do you have an I2C EEPROM soldered in?
Here another: Did you change the I2C2 bus speed? 
73
Danilo

db4ple, 2017-07-29:
Assuming you have no I2C EEPROM, it could be an interaction with the I2C bus do to the unsuccessful detection tries for the EEPROM on this bus.

Haasoperator, 2017-07-29:
Yes, I have an EEProm soldered in, and NOT the dinky little one originally called for in the BOM. I don't recall if I changed the I2C speed or not. I have the radio apart right now so I can look for bad solder joints, but I will post again once I have a chance to put it back together and test whether reducing the I2C bus speed(s) fixes the problem.
By the way, is the WM8731 codec necessary for the v. 0.6, or just an upgrade? And is it a direct replacement, or do mods need to be made to get it to work?
RichKC8MWG

db4ple, 2017-07-29:
Ok, you do have an I2C EEPROM. 

If you have changed the speed, you should check the speed and reset to default 100Khz. The newest firmware has this setting in the "Configuration Menu" towards the end of the menu items. That is the first thing you should do.

The WM8731 has been and is the audio codec of the mcHF from the beginning, i.e. all design versions use this chip (and the firmware supports only this chip). There is a TI TLV320AIC23(B) which is supposed to be fully compatible to the WM8731. 
In any case, I don't understand your question related to the WM8731 and ver 0.6.

DG9BFC, 2017-07-29:
maybe the two caps on i2c bus are installed??

db4ple, 2017-07-29:
But even with them, with the default speed I2C of 100kHz it should work.

Haasoperator, 2017-07-29:
One of the warnings I get on boot-up with the latest firmware is that the WM8731 codec is not detected. Which seems rather odd, considering the radio wouldn't function at all without the codec, regardless of the firmware used...would it? Or would the I2C speed being set incorrectly prevent the codec from being detected?

RichKC8MWG

db4ple, 2017-07-30:
The message is not a warning. It is an error message. And yes, the mcHF does not work properly if the WM8731 is not detected, since in this case no audio and TX/RX is possible. The user interface as such is not impacted by the missing codec, though.

No detection of the Codec does not mean it is not there, it means the firmware is not able to find it via I2C or to communicate reliably with it via I2C. So yes, the I2C speed setting can have a negative effect. 

73
Danilo

df8oe, 2017-08-04:
Hi Rich,

do you have solved your problem?

Haasoperator, 2017-08-04:
No, in fact after I took the radio out of its case to check for bad solder connections, I got an additional malfunction: the TCXO "window" on the display now has the word "SENSOR" in red, and the frequency numbers are in red as well. SO something else seems to have gotten screwed up as well. I sent the radio out to Chuck this afternoon; hopefully he can get it straightened out (and convert the TX/RX switching from PIN diodes to relay while he's got it apart for diagnosis...).
RichKC8MWG

df8oe, 2017-08-05:
Do you have tried reducing I2C speeds BEFORE you opened the case? Taht would be an interesting information, as Danilo and I recommended to test this more than one time....

The red SENSOR tells you, firmware cannot reliable communicate to temperature sensor IC via I2C1. This might be a problem with I2C speed, too.

73
Andreas

Haasoperator, 2017-08-05:
Yes, I did reduce the I2C speed beforehand, as low as it would go, and it didn't help. That's why I opened it up to look for bad solder joints. When Chuck rebuilt mine on 0.6 boards, he used some components salvaged from my 0.4, which also had some (different) problems, so it may have a defective component.

RichKC8MWG

df8oe, 2017-08-24:
This is no firmware issue but a hardware issue. Please close this issue.

closed 2017-08-24 by Haasoperator