df8oe/UHSDR #963: Saving configuration

sa4dht, 2017-07-29:
It would be great to be able to save the mchf configuration to a usb stick or via usb to a pc.

db4ple, 2017-07-30:
Please Andreas, leave this open and rename it to "Saving and Loading Configuration from USB", so that we don't get this newly entered all the time.

@sa4dht: This is a duplicate of #960 and some others.

df8oe, 2017-07-30:
This is already on our 2do list.

sa4dht, 2017-07-31:
Ok, great. As soon as I find some free time I will download the Eclipse IDE in order to be able to contribute with code changes as well. Normally I use Crossworks for ARM but I think it is easier to use the same IDE as you.

df8oe, 2017-07-31:
Reopened because of others must see this is already asked.

Lars-mcHF, 2017-09-11:
Thanks, I spoke before I found this

db4ple, 2021-02-19:
There is the uhsdr_tool to do just that. See https://github.com/df8oe/UHSDR/wiki/Backup-and-Restore

closed 2021-02-19 by db4ple