df8oe/UHSDR #964: in freedv mode the oad indicator and the "ber" value (plus freedv text) does overlap

DG9BFC, 2017-07-29:
title says it all ... load should be shifted upwards (below cmp box) in the same row as the agc and mode box

df8oe, 2017-07-30:
It cannot be shifted upwards because then it conflicts with RTC display. I think we simply remove the "load" indicator - there is no more space left on LCD.

db4ple, 2017-07-30:
Hello Andreas, please just disable with #if 0
I will find another way to place this, it is an important debug tool.

df8oe, 2017-07-30:
It is now conditionally enabled (if show_tp_coordinates is activated by pressing on "dB" on S-Meter when menu mode is active). We can also move freedv display to dsp-box because if digital modes are running DSP always is OFF - just an idea...

Fixed.

closed 2017-07-30 by df8oe