df8oe/UHSDR #966: PTT via RTS in CW mode

ok1cdj, 2017-08-01:
Please add possibility to control PTT via RTS in CW mode. Not so many SW are able to generate CW via DTR and PTT via CAT command. This feature add compatibility with many contest logs and also with cwdeamon in linux.

Thank you and 73 Ondra OK1CDJ

df8oe, 2017-08-01:
I think (PC) software designers should do their homework! Some SW wants to set rig to TX via PTT *AND* key CW additionally, some keys without first setting to TX, some are able to use CAT command for PTT (why not all? They do work with frequencey settings and so on - why not PTT?), some wants to use DTR, some wants RTS, some additionally are unable to use same serial port for CAT and DTR/RTS (different ports are not possible in firmware) - what a confusing thing!

I am happy that all software I use with mcHF is running perfectly with the now given features of firmware...

If we implement more, there must be configuration possiblities in menu. There will not exist one setting that works with all software.

73
Andreas

db4ple, 2017-08-01:
Hello Ondra, Andreas,

lets face reality: There is some software out there, which uses the RTS and DTR method (and not just one).
Until now every software we have been asked to support was able to do the CAT PTT & SERIAL DTR trick.
And now this is no longer true. I have been thinking about the added complexity and risk of RTS PTT functionality. There is a potential interference between CAT PTT and RTS PTT. However, only if the pc software uses both approaches at the same time which is unlikely. 

So in this special case, I would argue. we can add this without need to add additional menu configuration options. However, this functionality in practice interfers with previously working setups, we might have to remove this RTS support or make it optional.

ok1cdj, 2017-08-01:
hello all,  in real  CW  life is PTT over CAT realy slow and no usable for me. The combination of PTT over CAT and CW over RTS is not common in software (as i know it is possible only in FLDIGI, no in cwdeamon on linux (cwdeamon uses many kind of linux SW like TLF, CQRLOG), no in contest log  in windows like wintest).  I propose 2 ways how to solve this problem:

1) add CW VOX like in comercial rig. This mean only CW pin enough for key the RIG
2) add PTT via RTS

73 Ondra OK1CDJ

db4ple, 2017-08-01:
Hi Ondra,

PTT via RTS will NOT BE FASTER than CAT PTT in UHSDR. It is essentially the same approach. Alll CAT or RTS data is transmitted via USB anyway. BTW, the USB serial emulation speed in the UHSDR firmware is completely independent from the serial speed setting (it is always max possible speed).  

You should keep in mind, that there is some inherent software latency in the way the UHSDR handles the processing of a PTT request.  The two most important factors are the audio driver load "load" and the selected graphical visualization (none/scope/waterfall).

Only way to change this is a serious rewrite of the current code.

ok1cdj, 2017-08-01:
Hi, there is no problem on mcHF side. There is problem of delay in PC side. The CAT have no priority in contest logs. 

73 Ondra

db4ple, 2017-08-01:
I don't understand this comment. Why there is a delay on the PC side if CAT is being used? Same applies to RTS/DTR signalling I would say.

Anyway, there is contest software which does support DTR plus CAT (AFAIR): UCXLog and N1MM+
Yes, both run on Windows. But that might not be a problem for you. 
BTW, you should by now able to test RTS&DTR since the daily build has my changes.

db4ple, 2017-08-06:
Hi Ondra,
can you confirm RTS PTT working as expected?

TNX
Danilo

ok1cdj, 2017-08-07:
Hi Danilo, will test it today evening and let you know..

73 Ondra

S52CQ, 2017-08-10:
Hi Danilo, from which SW ver. onwards RTS PTT works?

73 de Jure

db4ple, 2017-08-10:
Hi Jure,
it is in since 2.5.29 but it did not work I heard yesterday. I just checked and saw the problem. Next daily build it should be working.

DG9BFC, 2017-08-10:
just a comment from a user who tried nearly all digital modes software (for windoze) since win 98 ... only a few of them support ptt via cat command ... but ALL DO WORK WITH RTS ptt ... be it sstv soft, psk31 soft, or whatever .. so that the mchf can be switched to tx via cat is nice ... but ptt via rts is a "must have" in my view ... only option if that does not work is a usb to serial adaptor with a transistor or optocoupler added and hard-wired to mchf ptt input (and then use mchf soundcard or the line in connector with pc soundcard) ...

db4ple, 2017-08-10:
PTT via RTS: Well it does work according to my experiment I just finished. Used N1MM+ to verify correct operation. 

Also fldigi was able to use PTT via RTS.

@Sigi: All programs I used had no problem without PTT via RTS, they could do it via CAT.
That almost twenty year old software doesn't support it is clear, but not sure if there not better alternatives out there: PSK31 is handled at least by fldigi (CAT PTT) and HRD (CAT PTT). I guess, we would find for each mode currently in use at least two programs which do use CAT. 
Not all things which can be done should be done without asking if this is really, really necessary.

db4ple, 2017-08-14:
Hi,
I noticed problems with the RTS as it is now: It works but some programs just set RTS when they startup, this put the mcHF into TX mode (fldigi AND wsjtx on Linux are like this).
I propose to disable RTS by default.  We will provide a menu item to turn it on for those who need it.

df8oe, 2017-08-15:
I can confirm PTT via RTS does make problems with many recent software so that it can kept OFF for those who does not need it...

db4ple, 2017-08-15:
Will commit today a fix for that (i.e. a menu entry to enable / disable RTS.

db4ple, 2017-08-16:
Hi, in the newest daily builds there is the switch to enable/disable RTS PTT in the Debug/Expert menu

df8oe, 2017-08-24:
This is implemented now. Because it impacts function of many recent HAM software it must be enabled in menu (and is not stored permanently). Please close this issue.

ok1cdj, 2017-08-24:
Thank you for implementation, this is ok for me.

closed 2017-08-24 by ok1cdj