df8oe/UHSDR #968: CW IQ adjustment

WD8BXS, 2017-08-02:
CW IQ adjustment does nothing, it works great for SSB, but I hear no change for the CW settings.

I am set up just like the calibration instructions say.
No traslation, etc...
What could be wrong???

Chuck WD8BXS

df8oe, 2017-08-03:
Do you watch at the correct offset frequency? Of course it is not 2*f(translate)...

73
Andreas

WD8BXS, 2017-08-03:
I follow this procedure:

CW TX 10m/80m calibration [from version 1.5.7]
Set USB mode. (Yes!)
Set Frequency Translation to OFF
You simply repeat all steps for the translated mode. The frequencies are: 
80m: dial frequency 3624.00 kHz, tune signal 3624.75 kHz, mirror signal 3623.25 kHz. 
10m : dial frequency 28.124.00 kHz, tune signal 28.124.75 kHz, mirror signal 28.123.25 kHz.

I hear the tones, but the adjustments have no effect.

Thanks,
Chuck

From: DF8OE 
Sent: Thursday, August 03, 2017 2:27 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] CW IQ adjustment (#968)

Do you watch at the correct offset frequency? Of course it is not 2*f(translate)...

73
Andreas

db4ple, 2017-08-03:
Hi,

I'd like to quote my own commit message from the commit which implemented CW IQ Adjustments:

"Untranslated/CW TX IQ Settings are now possible and applied if
CW TX or untranslated TX is happening.
Could not see much of an impact if using the CW TUNE function.
TEST IT."

Here is the link to the full commit for reference: https://github.com/df8oe/UHSDR/commit/a58c0c727d53947459f3d1ca7d9372e5a46053e4

WD8BXS, 2017-08-03:
Well, I read it but I don’t understand what you said.
Are you telling me that this adjustment does not work???

When I am transmitting on 10 meters, I have a small lobe on each side of carrier.
I was hoping that this adjustment would cure it.
Any ideas??

Chuck Wd8BXS

From: db4ple 
Sent: Thursday, August 03, 2017 8:02 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] CW IQ adjustment (#968)

Hi,

I'd like to quote my own commit message from the commit which implemented CW IQ Adjustments:

"Untranslated/CW TX IQ Settings are now possible and applied if
CW TX or untranslated TX is happening.
Could not see much of an impact if using the CW TUNE function.
TEST IT."

Here is the link to the full commit for reference: a58c0c7

db4ple, 2017-08-03:
Hi Chuck,
yes, I said, when I implemented it, the CW adjustments  did not seem to work as good as with the "normal" IQ adjustments. And I left it to the experts to figure out, why it is not working. The firmware applies the same approach as for the other IQ corrections (so it is not the code as such) but it is probably due to the fact the we are transmitting in non-translate mode. I am no expert in digital signal processing so I don't know why it is like this or how to fix this.

df8oe, 2017-08-05:
Everything is working technically as expected - I have confirmed it. Only description of alignment procedure was very bad. I have corrected it in WIKI. Please try again and report.

WD8BXS, 2017-08-06:
Just tried the new method, no difference

Chuck

From: DF8OE 
Sent: Saturday, August 05, 2017 3:12 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] CW IQ adjustment (#968)

Everything is working technically as expected - I have confirmed it. Only description of alignment procedure was very bad. I have corrected it in WIKI. Please try again and report.

df8oe, 2017-08-08:
Sorry - no idea. I have adjusted CW IQ (non translated) using this way in the last three days at 4 different mcHFs. Everything is working as expected. If you can adjust IQ at SSB (translated mode) it can only be a wrong alignment procedure - no software or hardware issue. That is my view...

WD8BXS, 2017-08-08:
Hi Andreas.

OK well let me tell you what I am doing.
I turn translation off.
I put the mchf on USB 3624, and my receiver on LSB 3624

Now I press tune (IS THIS correct?? or do I NEED TO USE A STRAIGHT KEY??)

Then I try to tune out the carrier with adjustments.
Then repeat for 28124

Is this correct??

Chuck

From: DF8OE 
Sent: Tuesday, August 08, 2017 4:14 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] CW IQ adjustment (#968)

Sorry - no idea. I have adjusted CW IQ (non translated) using this way in the last three days at 4 different mcHFs. Everything is working as expected. If you can adjust IQ at SSB (translated mode) it can only be a wrong alignment procedure - no software or hardware issue. That is my view..

df8oe, 2017-08-08:
Yes - that is correct. You do not need a straight key. Best is to misadjust TX IQ Balance/Phase (CW) first during alignment so that you can figure out what tone is coming up (and this is the tone you must suppress as good as possible).

df8oe, 2017-08-24:
Do you have got it? It is not a firmware issue - I have adjusted 7 mcHFs in the meantime...

WD8BXS, 2017-08-24:
Not yet, I have been very busy at Church this month. Almost finished with the project, then back to radios.

Thanks,
I will keep you posted.

Chuck WD8BXS

From: DF8OE 
Sent: Thursday, August 24, 2017 12:59 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] CW IQ adjustment (#968)

Do you have got it? It is not a firmware issue - I have adjusted 7 mcHFs in the meantime...

WD8BXS, 2017-08-25:
No luck here, I just tried a new one, but no difference in adjusting for CW. Nothing happens at all.

The translated one works just fine though.

Strange.

Chuck WD8BXS

From: DF8OE 
Sent: Thursday, August 24, 2017 12:59 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] CW IQ adjustment (#968)

Do you have got it? It is not a firmware issue - I have adjusted 7 mcHFs in the meantime...

df8oe, 2017-08-25:
Very strange. I only can imagine that you are watching the "wrong mirror frequencies". So I told you that you first misadjust phase and gain to the extremes (+ or -) so that unwanted signal gets maximized. Now it is much easier to identify.

In translated mode you do have many spurious signals with very low spans and you only can suppress ONE of them... Because of this suppressable signal may lay under the floor of the others (which you can not affect) it is much more difficult to identify it. And because of the spans are very small I cannot give concrete frequencies because of misalignments of your SI570 (or your RX) may point you to the wrong spurious...

Normally the suppressable spurious is already much better suppressed in non-translated mode than in trtanslated mode so you have missed it.

73
Andreas

df8oe, 2017-08-25:
Additional:
I will take a youtube in the next days that will show the procedure.

WD8BXS, 2017-08-25:
That would be great Andreas, I am sure it will help to watch you do it.

Thanks,
Chuck

From: DF8OE 
Sent: Friday, August 25, 2017 8:57 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] CW IQ adjustment (#968)

Additional:
I will take a youtube in the next days that will show the procedure.

WD8BXS, 2017-08-25:
What is the CW bias adjustment for??

Chuck

From: DF8OE 
Sent: Friday, August 25, 2017 8:50 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] CW IQ adjustment (#968)

Very strange. I only can imagine that you are watching the "wrong mirror frequencies". So I told you that you first misadjust phase and gain to the extremes (+ or -) so that unwanted signal gets maximized. Now it is much easier to identify.

In translated mode you do have many spurious signals with very low spans and you only can suppress ONE of them... Because of this suppressable signal may lay under the floor of the others (which you can not affect) it is much more difficult to identify it. And because of the spans are very small I cannot give concrete frequencies because of misalignments of your SI570 (or your RX) may point you to the wrong spurious...

Normally the suppressable spurious is already much better suppressed in non-translated mode than in trtanslated mode so you have missed it.

73
Andreas

db4ple, 2017-08-25:
Hi,
straight from Clint's KA7OEI Operator Manual which is a invaluable source of information regarding what many things in the firmware are about:

> CW PA Bias (If >0) - If this setting is greater than zero, it is sets the applied PA bias during
transmit when in CW mode. If this value is zero the setting of “PA Bias” (see below) is used.
PA Bias - This is the setting applied to the final transistors during transmit. If the value of
“CW PA Bias” is set to a value of zero, this value is used during CW transmit.
The signal gain of FET power transistors will vary with their bias. If you set the PA bias
in CW lower than that in SSB mode, you can expect that the RF power output will be
LOWER. The converse will be true if the PA bias is set higher (e.g. higher power
output.

I don't know if this makes sense (i.e. that anyone actually uses this) but here it is.

df8oe, 2017-08-25:
Hi Chuck,

please do *not* quote old messages because everyone can read them already here on GitHub and quoting may confuse...

In SSB you are TXing signals with different frequencies / different amplitudes. So your PA must be very linear - the best would be class A. Of course that is consumpting much power even if no signal is transmitted.... So you set bias to a reasonable height (so that it is not possible to frie eggs on mcHF) but not too low (so that your analog TX signal has good quality).

In CW you are only transmitting a carrier - one (hopefully) stable frequency at one amplitude. So your PA must not be so linear as in SSB. Many users are working in class C (== no bias). If you do want get more linear (you can reduce harmonics by getting more linear...) you can switch to C/B or B/A (increase bias setting so that idle just wants to start or starts only a few milliamperes). Most users rely that LPF eats harmonics so they do not set any CW bias...

vy 73
Andreas

df8oe, 2017-08-25:
Correct - Danilo...

Of course gain of MOSFET is increasing with higher bias. But my mcHFs do have so much gain that I must reduce power settings under all circumstances - so linearity is the point you must think about.

WD8BXS, 2017-09-05:
Andreas, 
Did you get to make the video of the IQ adjustment for CW yet??
73 my friend,
Chuck

WD8BXS, 2017-10-10:
I must say, that the last three that I calibrated worked just fine for the CW IQ adjustment.

WD8BXS, 2017-10-12:
Interesting observation, I get different power out using a straight key in CW mode than I do with the tune button.

Any ideas?

WD8BXS, 2017-10-12:
OOOPS didn't mean to hit the close button hihi

db4ple, 2017-10-12:
Tune in CW vs. Straight Key in CW ? Uses exactly the same code to generate the actual signal so this cannot explain the observed difference. Comparing CW Straight Key to SSB Tune -> different story, since different way to generate signal (CW: Direct IQ generation, SSB: Audio Signal, frequency shifted by 6 or 12 khz, filtered a bit and then transmitted -> here some difference can happen. How big is the difference you saw anyway? And how did you measure?

WD8BXS, 2017-10-13:
That is interesting!, I was seeing about 3/4 of the output on CW vs tune on SSB. 
Should I then use CW tune for all power adjustments??
Still don’t understand the use of the CW bias adjustment
No clear instructions anywhere for that.
When should it come into play??
Thanks,
Chuck

db4ple, 2017-10-13:
CW Bias: There was discussion related to that. We should have put the essenence of that discussion into the wiki's calibration page or FAQ page (or both). Anyway, AFAIR the outcome was, you can set the CW bias lower, to be more in a AB (reduces energy consumption) instead of A amplifier configuration. CW PA experts please help. If confirmed we should put a statement like this into the wiki. 

CW Tune vs. SSB Tune: It depends on what you want. We need to measure and verify the difference. If there is a clear difference between CW power and SSB tune power, we can adjust that easily if it is a systematic difference, not specific to your mcHFs. I would guess it is a systematic error due to different signal levels being finally generated. We'll see.  In this case a no-brainer.

df8oe, 2017-10-13:
CW Bias does not only have influence on linearity but enormous on PA gain, too. If you reduce PA Bias you reduce gain and so output power. So "output power" which is shown on LCD only is output power of the mode you do have adjusted it. If you adjust output power in SSB it is naturally not the same as in CW or AM.

It will be difficult to add an automatic correction for LCD display. If someone wants (not me) we can add a manual correction that everyone can adjust himself via menu. For me it ok if I knew output power at one mode (SSB or CW) and know relationship between the both. We all know it is without any sense to discuss if you are transmitting with 7W or 11W.

WD8BXS, 2017-10-13:
is OK since it is not continuous duty.
Thanks,
Chuck

WD8BXS, 2017-10-13:
I am not talking about the LCD display.. I use a QRP wattmeter.
I have a radio here on the bench that just came in.
It has full 5 watts out on QRP wattmeter, on SSB.
When I key in CW I only see 2.5 watts.
I have observed, that if I use 2 tone tuning on SSB and set the power out in the PA menu then the CW also has the same power out.
Interesting. Don’t you think?
Chuck

WD8BXS, 2017-10-22:
Thanks for all of your help!

closed 2017-10-22 by WD8BXS