df8oe/UHSDR #970: SideTone volume not linear

wk8s, 2017-08-03:
Installed latest firmware from Thursday's Daily Snapshot 8/2/2017 - Volume as adjusted in STG/AF gain was either to low at setting 1 or 2 or jumps to extremely loud at setting 3 and above. Not linear at all. I went back to 2.5.11 and the side tone was again linearly adjustable.

db4ple, 2017-08-03:
Yes, this was a recently introduced regression, we fixed that, wait for the next daily build.

wk8s, 2017-08-03:
Thanks - no worries. 

Pete Meier
pmeier@me.com
Amateur Radio Callsign: WK8S
|| I never lose. If I don't win, I learn. - unknown author ||

s53dz, 2017-08-04:
I suppose both AF and the STG gain should follow the power-logarithmic rule to be near the human ear's perception and not the voltage-linear one.

73 Bojan

db4ple, 2017-08-04:
The Codec internally maps linear values to a decibel-based volume change

db4ple, 2017-08-06:
Pete, if the issue is gone, please close it.

TNX, Danilo

s53dz, 2017-08-06:
Just a report,

For me, in FW 2.5.32 the issue is gone.
TNX 73 Bojan

wk8s, 2017-08-06:
The issue is resolved- thank you

closed 2017-08-06 by wk8s