df8oe/UHSDR #972: small graphic bug

DG9BFC, 2017-08-04:
when switchin on freedv mode the text "ber" flickers ... and when switching off freedv there is some "rubbish" left over on the screen (screen is not black in that area but has some yellow text fragments left over)

db4ple, 2017-08-04:
Yes, that is due to the better support of line breaks. Obviously, sometimes FreeDV reports "linebreaks" and starts writing in "forbidden areas".

No very urgent, but annoying.

DG9BFC, 2017-08-04:
oh ... and when you let freedv run for a while it overwrites the frequency readout (vfo numbers) ... THAT is really annoying

df8oe, 2017-08-05:
Switching off large VFO digits is a bad option because small digits contains RIT information. We do not have enough space on LCD to solve this at the moment. I do not see alternatives which do not impact other functions. I think we must live with it - I do not have any problems with it ;)

73
Andreas

db4ple, 2017-08-10:
Hi Sigi,
please check if it is now working for you. Have currently no ability to run a FreeDV QSO easily.

73
Danilo

DG9BFC, 2017-08-10:
i just let bit run for a while (a few minutes) and the noise gave enough "rubbish" to show up in the freedv text line ... just testing newest daily and the freedv text fragments after change mode/band etc ... are gone
so ... the text rubbish is cleared when you change mode .. so seems to work
if there is still rubbish text writing inside the vfo numbers is what i could not test yet ... but in a few minutes it will be better to say something about that ... (it takes a while with only noise on the frequency that the text box is filled up with some text)

DG9BFC, 2017-08-10:
ok ... it seems to work fine now ... freedv does not write in vfo numbers .... the text is a scrolling line now and there is no line feed disturbing the vfo numbers ... and "ber" does not flicker (or exactly is does flicker only till the first letter is written in the freedv text line ... then the text ber is solid)

closed 2017-08-10 by DG9BFC