df8oe/UHSDR #973: In Scope Mode (not light) close to top of carrier line  small rectangle appears

db4ple, 2017-08-05:
We had a report for that already. I now saw the same. I had set grid color to magenta, center carrier line color to green. And suddenly a structure appeared, it has different shapes but always the color of the center carrier line.

df8oe, 2017-08-06:
That must be dependent to something else. I have tried same settings - nothing changes... No structure, no little square...

vy 73
Andreas

db4ple, 2017-08-06:
It takes a while, I left it running for some hours. It seems that this happens when some special conditions appear (which seems to happen randomly)

db4ple, 2017-08-09:
Code has been cleaned and rewritten. Haven't seen the problem, but it should be gone, did not reappear after code rewrite.

closed 2017-08-09 by db4ple