df8oe/UHSDR #986: new noiseblanker produces "hum noise"

DG9BFC, 2017-08-13:
title says all
if i switch on the new nb (on a free frequency so it can be heard easy) i hear some hum noise
that seems only to be present in am mode and if i use low pass filters in ssb mode (in bpf filters that hum noise seems to be surpressed) ... in sync am i also can not hear it (or it is burried in the noise)

so .. to hear that hum noise ... switch to am on a clear frequency (or use a lowpass filter in ssb) and switch the new noiseblanker on and off .... i can clearly hear that hum noise coming out of my speaker

i have some electric fences here and i can say that the new noise blanker does not help much (not much better then the old one) ... only thing that does help a bit is the new agc (wdsp agc) ... but i tried all combinations of agc and nb settings and can not surpress the fences noise :-(

df8oe, 2017-08-15:
I cannot confirm using my working conditions. My "electricfence" which is probably 300m away vanishes completely if I turn on new noise blanker and there is no hum noise.

Lars-mcHF, 2017-09-11:
Is there a way to export, and then email, and then import settings such as freq, DSP BW & Mode to replicate settings to help identify and discover problems and solutions?

df8oe, 2017-11-04:
Is issue gone? if so: pse close it.

DG9BFC, 2017-11-12:
no issue is still present
but i run display in parallel mode ... and the hum noise seems to be dependant on what you see on the screen
so maybe a side effect of the "helocopter noise" ?!?

DG9BFC, 2017-11-29:
issue seems to be solved ... i can not hear that noise anymore

closed 2017-11-29 by DG9BFC