df8oe/UHSDR #988: TX/RX switching time too long in SSB

DF5LI, 2017-08-14:
The title says it all.  In a SSB-QSO the time from TX to Rx is so long that you will lose the first syllables from your qso-partner.  In CW there is no delay. Could that be fixed?
73 Harri

db4ple, 2017-08-15:
Well I found 2 places now, one in the PTT handling, which is probably best left intact in a way as it handles bouncing but we have another delay, ts.audio_spkr_unmute_delay_count, which is set to the equivalent of 300ms and which probably causes the "long" delay. 

This is not and should not be a user changeable parameter as far as I can set it. I think it was purely motivated by technical needs such as avoiding unwanted sounds from the RX until is has been feed a little audio for processing. I doubt that we need to do this for 300ms, knowing that this delay happens only if we have a microphone as input. We will check and see what delay is appropriate (as little as possible). 

73
Danilo

df8oe, 2017-08-15:
Hmmmm. I am working in SSB using mcHF every day. I never have had problems with the existing delay. I do won some commercial TRX which do have a slightly longer delay... So I thought that this is technically neccessary and don't have thought about it any time.

73
Andreas

pa5ca, 2017-08-16:
Hi, I'm PA5CA, I just have completed my mcHF TRX and made my first QSO on 40m SSB. I also experienced the TX hang phenomenon .. it takes about 0.5.. 1 second before the RX becomes available.
I use a of the shelf hand held microphone, the only modification was the stereo plug.

yours 
73 Frans PA5CA

closed 2017-08-20 by df8oe