df8oe/UHSDR #992: Does not powerup while USB is connected to PC (Windows 10)

sm0tsc, 2017-08-15:
Only white screen is shown

73 de SM0TSC

db4ple, 2017-08-15:
Hi,

Firmware Version?
Bootloader Version?
Hardware Version?

Did this happen with earlier firmware/bootloader? If not, which was the last firmware which you used?

TNX,
Danilo

sm0tsc, 2017-08-15:
Firmware 2.5.46 / Bootloader 3.2.1 

It did happen with 2.4.0 aswell

//SM0TSC

db4ple, 2017-08-15:
Ok. So it is not a new thing for you. 2.4.0 has been around for a while, if this would be widespread problem, we should know. So in your case is something unique. No idea what, but lets check:

- What kind of PC (windows [version], Linux, Mac)? 
- Did you try a different computer and had the same problem?
- Once powered up with USB cable disconnected, does USB work if you connect it after sucessful boot? PC should at least detect mcHF audio and serial USB devices.

sm0tsc, 2017-08-15:
I am completely new to mcHF so 2.4.0 the first I loaded. Then trying today with latest daily. Same thing.. Using PC with Windows 10 also tried another PC (Also win10) with same result.. If powered up without being connected to PC then connect to PC everything works fine

db4ple, 2017-08-15:
Ok, that means, you do have a working setup (so this is not extra urgent). I can confirm that normally having Windows 10 connected to mcHF during boot should not cause trouble, works for me.
So we have to think what it could be. Might take a while.

sm0tsc, 2017-08-15:
Nothing urgent. Just wanted you to know...

//SM0TSC

s53dz, 2017-08-16:
Hi,

did you try connecting a battery-powered laptop (no adapter connected)? Perhaps this is not Win10 issue at all.

73 Bojan

sm0tsc, 2017-08-16:
Does not matter if laptop i battery or AC powered... Does not work.. BUT if PC is turned off when powering up mcHF or disconnected from USB it powers up correctly

73 SM0TSC - Johan

sm0tsc, 2017-08-16:
We also tested on a friends radio, and it behaves same..

73 SM0TSC - Johan

df8oe, 2017-08-16:
I cannot confirm. All my mcHF's are powering up normal if connected to my PCs (all running Linux). First we must test if it is a problem regarding Windows - or a problem regarding your PC.

Please set your PC to BIOS (or stop normal booting before Windows has loaded - I do not know if this is possible at recent Windows versions - I remember there was a "F8"-Option many years before).

If your PC is ON but Windows has not loaded yet connect it to your mcHF and try again. If it is now powering up ==> Windows problem ;)

You can test using another PC or Laptop, too to see if it is hardware related to computer.

73
Andreas

db4ple, 2017-08-16:
Windows 10 + mcHF connected via USB works for me.  But testing with just the BIOS is a good idea, although also the BIOS may also initialize USB. 

May problem with that issue is, that we are starting our "USB" driver only AFTER the display has been initialized (no white screen). Before that only the STM HAL is working but no device is up and running.

You did not report the hardware version. I assume 0.6:  Have you fitted R43b? If so, remove it and try again.

df8oe, 2017-08-24:
Any new observations here? I do not think it is a firmware-related issue...

sm0tsc, 2017-08-24:
Nor do I I have also updated both bootloader and firmware.. I will try the
resistor removal when I got some spare time...

//TSC

df8oe, 2017-11-04:
is issue solved? If so: pse close thread.

isenburg, 2017-11-14:
same thing here with Firmware version 2.5.130

db4ple, 2017-11-14:
Hi Isenburg: Also for you the question: You did not report the hardware version. I assume 0.6: Have you fitted R43b? If so, remove it and try again.

73, Danilo

Osrosado, 2017-11-21:
I have two cables for the DFU one does the white screen the other does not work normally. I dont believe I have a bad cable but perhaps there are cables that are wired differently or for a specific device?  If this is a clue. One cable powers and charges my Uniden portable that same cable causes the white screen on the UHDSR. The other cable does not power/charge my portable but it works on the UHSDR normally.  Both cables, however, work on the UHSDR for data/PC one needing to plug in after bootup. 

I hope this helps in someway it maybe your cables.

isenburg, 2017-11-21:
Hi, 

I do have H/W version 0.6. Unfortunately I haven’t had the time to open it up again to check. I will try different cables first and keep you posted.

regards

Georg, DJ6GI

isenburg, 2017-11-21:
thanks for the tipp it was the cable...

df8oe, 2017-11-21:
Very interesting. I never would have thought there are cables which are leading to this issue... I think this isue can be closed now?

closed 2018-01-30 by df8oe