df8oe/UHSDR #994: filter switching

mcafzap, 2017-08-15:
I note that the relays are only set when changing between the bands using Band+ and - buttons. Nothing happens when changing between the higher frequency bands; so the question is, should the relays be set again when this happens? Similarly when tuning up or down and going into the adjacent band; should the relays be set then too?

df8oe, 2017-08-16:
Regarding the schematics 15/17/12/10 are using all the same filters so no switching if you change between these bands/frequencies. Please take a look at the schematics (RF board, page "LPF"). Of course filter switching is bound to frequency changing - not only to BAND+ and BAND- !

mcafzap, 2017-08-16:
What I was suggesting is that despite all these bands using the same filter, it might be a good idea to re-send the 'relay commands’ for every band change just to make sure the filters remain correctly set.

db4ple, 2017-08-16:
Hello,
honestly, why should we do this? There has never been a report to indicate the switching to be a problem in a working mcHF. Do you have an issue, i.e. switching not working reliably?   

The software intially brings the relays into a known state at startup and from that state, it switches when a new configuration is required. This must work in the first place anyway, so additional "making sure" should  not be required. If it did not work, switch to a different band and do it again.

mcafzap, 2017-08-16:
Fair enough. (In my experience and particularly around large amounts of radiated power, microprocessors can lock-up etc.)

df8oe, 2017-08-16:
We never have had any issues regarded filter switching. If you power on mcHF, filters are set. If you change frequency (regardless if via VFO knob or CAT) filters are set. This includes frequency changes via "band change". No need to re-set relays. Not one single issue report all the years.

db4ple, 2018-05-16:
Hi, 
which firmware version "does" work? And what hardware are you using? mcHF version ? Built from a M0NKA kit or did you source the parts by yourself. Most likely you don't use the "normal" relays but some with other specs (which need a longer pulse to switch). It might be the voltage (the right ones have 3.3 volts).
Since the UHSDR firmware works for most users, it is very, very likely a "hardware" relaled problem.

73
Danilo

db4ple, 2018-05-17:
That might be the case. If this is the cause, you may press F1, F3 and F5 during power up. This allows you to reset all configurations to their defaults. If this helps (I don't know why it should but let us try this), just power off normally. If it doesn't help, power off by pulling the power (in this case the old configuration remains stored in EERPOM).

73
Danilo

anivalex, 2018-05-17:
Thanks. I will try it evening and write here result.

anivalex, 2018-06-13:
Hi all.
I decided the problem. Problem is in BAND2 signal. Active is low, but UHSDR set it HIGH. I decide problem with remove R30, and connect pin to U14 to GND. If you think, that is need to correct this signal in source code and next version on UHSDR. 
Thanks!

df8oe, 2018-06-13:
Very stange. It is working on all of my mcHFs and all mcHFs of all friends. Additionally there is no other issue report so I cannot believe there is something wrong in firmware regarding filter switching. You know that BAND signals are no constant voltages but pulses??

df8oe, 2018-06-13:
We need more confirmations. I cannot reproduce the issue on any radio.

anivalex, 2018-06-13:
Please view LPF circuit. In there have table for signals. mcHF_rf-v0.6.pdf - page 4. How you inited BAND2 signal via firmware, LOW or HIGH?

df8oe, 2018-06-13:
It must be initialized correct - otherwise it wold not work on any mcHF. But it does. On hundreds - and on hundreds clones, too ;)

So it must be hardware dependent - no other possibility. The used initialization is working!

Critical can be if relay's switching speed is too low. If so pulse may be too short to switch. But in this case used relays are not the ones which are recommended...

Let us investigate where the problem is located.

First is to see if bandpass filters are switched correctly. You can measure DC resistance of switched paths in U1 or U2. Please take a look if they are switched correctly.

anivalex, 2018-06-13:
U1 and U2 is BPF, not LPF. BPF is works correct, because is not use BAND2 signal. LPF only use it, but I not understand why. The signal is must be only LOW level to do works all. If band is switched, BAND2 signal is switch impulse to LOW for short time, and after that is return to HIGH.

df8oe, 2018-06-13:
The difference is that relays are bistable versions. You only must set a switch pulse and then they keep their state. There is not any logical solution that can explain working and non-working simultaneously except the switching time which is of course critical. You can see length of pulses if you connect an oscilloscope in parallel to relay coils.

anivalex, 2018-06-13:
My relays is simple relays. Have voltage relay is on, not have - relay is off. This is original relays from Kris circuit.

df8oe, 2018-06-13:
This is wrong relay. You need bistable ones. Please compare part numbers to BOM.

db4ple, 2018-06-13:
Hi,
unfortunately you have to remove your "simple" relays and put the correct bistable (i.e. latching) version in. Chris, M0NKA specifies G6KU-2FY 3DC in the mcHF 0.4 rf schematics (and all subsequent ones):
http://www.m0nka.co.uk/wp-content/uploads/2013/12/rf__0_4_schem.pdf

Learn more about the relays for instance here: 
http://uk.farnell.com/omron-electronic-components/g6ku-2fy-3dc/relay-signal-dpdt-30vdc-1a/dp/1181070

Please close this issue if you are not using the above mentioned relays or relays with equal specifications.

Thank you,
Danilo

anivalex, 2018-06-13:
My relays is exactly this: G6KU-2F-Y 3VDC. But I think, relays is not bistabile. I bought it from TME.

df8oe, 2018-06-14:
Two possibilities:
1) Relays are monostable. In this case: swap them to bistable version because only this is supported.
2) Relays are bistable

In case if 2) you have to investigate the hardware issue at your board which leads to not switching. It is definitely not a firmware problem.

anivalex, 2018-07-27:
Hi all again!
I investigate the problem, and not found decide for it.
- Via original firmware impulse on BAND2 pin is two time long that UHSDR firmware.
- Via original firmware, enable signal on BAND2 is from 1 to 0 twice.
Please tell me how I extend impulse more time?

I underline, with original firmware all is works normaly! Relays is switch normaly!

If I fix BAND2 input to permanent Vcc bands from 1.8 to 14 works normaly,
but over 14 relays not switch on fourth filter, and stayed on third.

Help me!

df8oe, 2018-07-27:
If your hardware is identical to the "original mcHF" hardware all will work fine. There is no firmware related issue in this section so I can only repeat: rework your hardware. There is definitley no other user who has problems here.

anivalex, 2018-07-27:
Do you possible to compile special version for me with two time long impulse to I test it?

df8oe, 2018-07-27:
Sorry - no time...

db4ple, 2021-02-19:
Closed due to inactivity

closed 2021-02-19 by db4ple