df8oe/UHSDR #997: 5W adjustment

pa5ca, 2017-08-16:
preforming the 5W calibration (adjustment) it came to my knowledge that the mcHF switches to full power, can be observed in the power box on the display. At this moment you are not able to set the 5w gain..
The work around, if you have pressed the tune button to press the power button in such way that the power pox on the display shows 5W. From this moment you are able to set the 5W gain.

I'm not sure if this is an known issue I just like to report it. In the mean while I have add some information on the PA calibration page.

yours,
Frans...

db4ple, 2017-08-16:
I think, we can close this now, what do you think, Frans? That would be your job then.

73
Danilo

pa5ca, 2017-08-16:
Hi Danilo, I still have doubts about you text modification..setting the power to 5w in step 3 wiil not work.. thats why I wrote to do it while the tune is active!!! 
If you set it at step 3 and you press tune the tx power jumps back to FULL power..
So we leave the issue open..:-))

df8oe, 2017-08-16:
Hi Frans,

the possibility to set TUNE POWER as a user-defined value was introduced about 1.5 years ago and for all who do posess an mcHF do know this...

But now description is absolutely clear - I think. If you set TUNE POWER AS TX POWER everything will work as expected. I think we also we can close this... I do not see any issue.

73
Andreas

df8oe, 2017-08-16:
Hi Frans,

I have tried the following two possibilities:

1) Setting TUNE POWER AS TX POWER
and
2) Setting TUNE POWER to 5W

In case 1) you have (of course) first to set TX power with button G3 to 5W, in case 2) there is nothing special to do.

Just press "TUNE" and you are TXing with 5W and you can adjust power levels for each band. Absolutely un-misunderstandable and easy to do... Of course you must first select 5W output before you can adjust it - that's now in the description.

73
Andreas

pa5ca, 2017-08-16:
Give it a try this evening... thanks for the comments... and assist keep you posted...
I'll close the the issue..

73 Frans

closed 2017-08-16 by pa5ca