df8oe/UHSDR #1004: --CW DECODER FW D2.5.54 from 2017-08-17

DL3AAA, 2017-08-17:
Hello, dear developers, 
if the possibility passes the CW code to extend still by 3 german umlaute, then please the following insert: 
ä =. _. _ 
ö =---. 
ü =.. -- 

Thanks 
DL3AAA 
Dietrich

phaethon, 2017-08-17:
For decoder it is easy (corresponding codes are 187, 254, 175, which have to be added into array in cw_gen.c).
But as these are not ASCII characters somebody should confirm that it does not break anything in the function printing the characters.

db4ple, 2017-08-17:
There are no umlauts in our character set (ASCII only). Of course we could change this, but this opens a can of worms. What about icelandic, danisch, swedish, russian, you name it characters? If someone volunteers no problem...

df8oe, 2017-08-18:
I think I do have a solution for this. It is not a problem to add a table for specific prefixes (selectable via menu) and pointing to special CW characters. Decoder can identify / understand it. That is the main wish of Dietrich. The output on the screen can be "ae" in case of "ä". And I think there can be an ascii combination of EVERY special character. The table cannot be complete or perfect at the start. But it can grown in time. And it will not need much flash size and give CW users the ability to work in CW using their country specific CW characters. It is a benefit.

db4ple, 2017-08-18:
Hi Erik, +1 for this idea. We should probably have binary search instead of linear then to limit the number of checks. 
BTW, I would propose to change the data structure a little (this would give us a single table as long as the morse code is unique):
```
struct cw_character 
{
  uint16_t code;
  char*      value;
}

struct cw_special_character cw_special_char_map[] = 
{
  { 756, "a" },
  { 2341, "b" },
  { 4341, "ae" },
}

#define CW_MAP_SIZE  (sizeof(cw_char_map)/sizeof(cw_character))
```
This way CW_MAP_SIZE is always in sync, numbers always match code. And no problem with match code and value. We can store single character strings and also longer strings for umlauts etc.

For the evolution of the table this has a nice property, since you just add a single entry in the right position, which is not hard to find. The price is a little more data (5 bytes per character but this amounts, by using uint16_t instead of int (which is int32_t) we save 2 bytes, so there is acceptable overhead.

73

phaethon, 2017-08-18:
I agree that the structure improves status quo, but I think there are more issues.

There are, also, prosigns. 
Also, there could be a reason to translate into another direction (which would be impossible if we store it as "ae"), e.g. to send stored morse message. 
Having multiple bytes per character could be issue for buffered text processing when it will be used for later transmission over PSK or RTTY. Storing "ae" in the string will result in erroneous transmission or a nasty code to differentiate.
There may be need to have different code pages (e.g. as you mentioned cyrillic). And it may not be good idea to load tables of all code pages (and fonts) into firmware (because of space usage). One way could be checking for and reading font and codepage description when loading firmware from flash and storing into the EEPROM. Another benefit would be that we could use one byte per character in the string representation, which would simplify the processing code.

To summarize, I would not love having multiple bytes per character at this stage. If we want to support some non-ASCII characters right away, we can add them in the 'old codepage style' keeping character representation to one byte. For displaying them - either add a character to the font, or write a small translator (e.g. translating one character to "ae") within the text displaying part. In a longer perspective it will likely need to be significantly refactor anyway.

db4ple, 2017-08-18:
Erik, you got me wrong: The translation using the strings I suggested was only for screen presentation. For editing, I completely agree, it does not make sense to use translated characters. 
However, you could actually use the morse code itself as symbols or a mapping to unicode (UTF-16). Both could be stored in the same table. This would make it easy connect a keyboard, which will have just another mapping.  But I agree, this is not an urgent topic.

db4ple, 2017-08-18:
BTW, the decoder is really cool stuff. I finally built a 3D-printed paddle from thingiverse and started to send morse (well did not really send it out) but anyway. It think this will make a create tool for practicing morse code, maybe at one day even I will be able to send something out in CW. Or even read CW. Thanks, Erik!

phaethon, 2017-08-19:
Thanks @db4ple for the good words!
Regarding international characters, let's give it some time to settle down the ideas. Prosigns need some form of translation, too, so we will come back to this anyway.

closed 2018-01-30 by df8oe