df8oe/UHSDR #1009: bootloader 3.4.1 problem

S54AM, 2017-08-19:
Hello Andreas,

I am affraid that is somthing wrong with new bootloader edition 3.4.1. After upgrading radio does not work, just grey LCD apears when power button is pressed. The old one 3.4.0 works fine.

df8oe, 2017-08-19:
Very bad. I only rebuilded the one from July, 1st without testing (and do not have any possibility to test at the moment) :( Nothing changed. We will take a look - thank you for your report!

df8oe, 2017-08-19:
Please can you add information about your LCD: is it parallel or SPI?

S54AM, 2017-08-19:
I have SPI LCD. If you need someone for testing your new BL, I am here.

df8oe, 2017-08-20:
My newly build binaries are working - it was a problem during UPLOAD :/ This was not detected by my automation. Now I populate md5sums for each file and automation tests after upload if everything was successful. Users can check their downloads using the md5sums, too.

Files are pushed again to server - pse check and close if it is working for you now.

S54AM, 2017-08-20:
Now everything is OK.

closed 2017-08-20 by S54AM