df8oe/UHSDR #1011: External keyer or straight key

ve3ids, 2017-08-21:
I am having problems trying to use an external keyer or straight key. The internal keyer works great. The start of the character is clipped each time the mcHF goes to TX mode. If the txrx delay is increased, it will send ok as long as the delay is high enough so it doesn't drop between characters. At a word space it will mess up the first letter again. I have a version 6 with the August 19 daily build. 
Thanks
Don ve3ids

la2fda, 2017-08-21:
Like
https://github.com/df8oe/UHSDR/issues/659 and https://github.com/df8oe/UHSDR/issues/669 ?
I think you find a solution here.
73
Stein

db4ple, 2017-08-21:
Hello Don,
according to my measurements when working at #669 we had about 23ms delay going from RX to TX in CW with straight key. There may be additional delay before we start processing the first character since we have to wait for the main UI to decide to go to TX. This delay depends a) interrupt load (filtering & DSP use) and the waterfall / scope settings. Please disable waterfall or scope ("increasing" speed to 0 will disable scope or waterfall) and test if this provides improvements.

73
Danilo

ve3ids, 2017-08-21:
Hi Danilo and Stein,
Thanks for the info. That is the exact problem I am seeing. Disabling the bandscope and waterfall does help. External keying will work up to about 15 WPM now. Above that and it still messes up. I am assuming this is a processor loading issue that the small MCU can't handle? I tried increasing the TX initial mute delay and that made it much worse. I can send slowly with the hand key with the display turned off, it would be nice to have high speed external keying but I understand the hardware limitations. This would be a good feature to consider in the new design. 
73 Don ve3ids

db4ple, 2017-08-23:
Hi Don,
could you so kind to rename the issue to be a little more descriptive (I cannot do this). I would recommend to "CW Mode: start of first character clipped with external keyer or straight key" just to let the others know what this is about.

The problem is not directly a hardware problem, the SMT32F4 is absolutely capable of responding much faster, it is a problem of the current software architecture. I have an idea how to solve this but it is not trivial to implement so that it works without causing problems.

So stay tuned.
73
Danilo

ve3ids, 2017-08-23:
Thanks Danilo for your interest. I would like to have a solution to the problem and I realize it will be considerable work. I am really happy with all the other features of the rig and am amazed at the great features you fellows have been able to incorporate into it. I am enjoying it with the fine internal keyer in the meantime. Thank you for all the hard work that has been put into it.
72/73 Don VE3IDS

s53dz, 2017-09-10:
Just for information about the CW dot values to help further:
An example: 
at CW with WPM=20 and weight=1, a dot duration is 60 ms (which is slow) but
at CW with WPM=28 and weight=1, a dot duration is 44 ms.
So, I think for a normal CW the delay could be something like 10 % at the most.

73 Bojan

flesnick, 2017-09-26:
I too am having the same issue of the first character being cut off if Inallow the break in to open or leave the space too long that the radio does break in. Have listened tonit on another radio and can confirm. And this is with a straight key and not the keyer 
Fred VE3FAL

df8oe, 2017-09-26:
We are reaching the edge of this little radio. I am not sure this can be fixed either because RX/TX switching needs a minimum of time (LO must be retuned, code for calculating sine waves must start and get "clean" output (first ms must be clipped).

So are is only workarounds: use the internal keyer or add a delay so that full BK is disabled. Using internal keyer we can delay the start without clipping the first dot reasonable.

I do not think mcHF will ever become a "high-speed CW TRX" and it is not yet constructed to reach this aim...

But if we are working on audio path maybe there are some ms to gain. We will see... Priority of this is not high because using internal keyer all is working well.

73
Andreas, DF8OE

KS4L, 2017-09-26:
I have just discovered that my mcHF (0.6 boards, 2.5.72 firmware) also has this problem.  I got the radio for CW, including straight key work for skcc, so this is a major disappointment for me.

73,
Randy, KS4L

df8oe, 2017-09-26:
ALL mcHF s do have this "problem". The newer firmwares are *much* faster than the old ones so we already have made a huge improvement. But we are getting into regions where a few ms are disturbing highspeed - CW - but a few ms are definitely needed by software for getting from RX to TX, for switching relay from RX to TX ( and relay is MANDATORY because of pin diodes leads to very distroted TX signals). Additional to get rid of the "popp" at each TX start we must (!!) add a few ms of muted signal to every TX start.

Randy, I think you have choosen the wrong TRX for your working conditions... There are better ones - preferred no-SDRs...

If I am working in CW I use maximum speed of 18...20WM which is the maximum speed I can handle. At this speed I cannot see any issues.

73 de
Andreas, DF8OE

ve3ids, 2017-09-26:
Andreas, will the OVI40 have straight key operation or external keyer operation? I would like to build the OVI40 as well but if it has this issue like the mcHF has, it won't do what I want it to do. 

73 Don VE3IDS

db4ple, 2017-09-26:
Hi,
please, do not speculate wildly, but read my comments in the upper part. The issue is related to the way the current software is written. It is not the speed of the processor as such, the STM32F4 is more than capable of doing CW first dit with an external keyer perfectly. But it requires a major rewrite of certain parts of the software. Consequently the OVI40 has the same problem as it shares most of the software with all other UHSDR devices (aka the mcHF).

73
Danilo

KS4L, 2017-09-26:
In my opinion, any QRP transceiver that won't do a good job at CW is doomed.

73,
Randy, KS4L

df8oe, 2017-09-27:
Hi Randy,

that is your conclusion drawn by your position of view. In my opinion UHSDR is already doing a good job in CW. Not a perfect one - all right. But you can work in CW with internal keyer with mainly any problems and if you want to use straight key (or external keyer) you only have to leave full bk or reduce the speed to speeds most CW QSOs do have.

There will come up the time we will completely redraw all audio functions and paths because the existing ones are not flexible and extendable in the way we want. This is a point where we will think about other structures and hopefully CW speed will rise up the speed which technically cannot be improved.

CW routines in radio are ~5% of code - we must stay carefully to not make radio unusable due to sporadic firmware crashes because of routines are left fast but "unclean".

73
Andreas, DF8OE

df8oe, 2018-01-30:
tested with speeds up to 30WPM: no more reproduceable. At higher speeds may be.

closed 2018-01-30 by df8oe

KS4L, 2018-01-30:
Andreas,

Checked briefly, and looks good! With a straight key or bug I send around 25 wpm and works well at that speed! Don’t know what was done, but I really appreciate it!

73,
Randy, KS4L

ve3ids, 2018-01-31:
I just installed 2.7.76 and I still get the first dit clipped about half
the time. It seems to happen with some sort of timing. If I send the dit at
the right moment, it is ok, but if it starts when the rig isn't ready, it
misses it, if that makes sense. Do I need some other settings applied?
If I hold the external keyer dit paddle, it will miss about 90 % of the
dits and turn them into dash length characters. By setting the tx rx delay
up so it doesn't drop out, it will work in non QSK mode but still misses
about half of the first dit of the sequence.

Thanks
Don ve3ids

df8oe, 2018-01-31:
I cannot confirm and many others not, too. So I think it is configuration dependent or it is a hardware issue. Additionally it is not possible to improve there something without **many, many, many** work. So actually nobody will start there anything.

Again we are missing a test report from those who do have this issue, if disbaling waterfall/scope changes the behaviour.

73
Andreas

ve3ids, 2018-02-01:
I have found that turning off the LCD will allow the external keying to be
stable up to 26 WPM. My workaround for now is to set the LCD timeout for 2
seconds. Also waterfall mode is slightly better than scope mode when the
LCD is on but not by much. The only change that seemed to help was to kill
the display, I tried many other settings with no changes to the issue. I
could not find a way to shut off the scope only and leave the rest of the
LCD active for a test  I think there was that option at one time but maybe
there isn't now, I couldn't find it.

73 Don ve3ids

db4ple, 2018-02-03:
Please try if the next (!) build (2.7.88 or later) will reduce the issue. We added dropping out from display update (which is the most time consuming activity during CW). Please report your findings.

ve3ids, 2018-02-04:
I have just install 2.7.89 and it works great up to 30 WPM. Beyond 30 it
will misbehave a little but still not too bad up to 35. I added 20 ms RX TX
delay and it helps and isn't noticable, it is essentially QSK with that
small delay. 30 is plenty fast enough for me, I'm a happy camper!

Thanks very much for this great improvement, it is greatly appreciated!!!!

73
Don ve3ids

df8oe, 2018-02-04:
Thanks for aour report! I think we would have started that much earlier _if someone would have posted test results w/o activated scope/wf_.... But nobody has tried this before - so it seems not very urgent...

db4ple, 2018-02-04:
Yes, I mentioned that this should be tested on Aug 21th 2017 in comment number 3 for this issue... can barely remember that. But github almost never forgets a thing..
Anyway, I am happy that we could improve this with a simple change (effectively 2 lines modified). 
And to be clear, it is not a general cure for the underlying fundamental problem.

KS4L, 2018-02-04:
Happy to report that here CW is working great with straight key input 
with keying speeds as high as I can send (probably around 25 wpm).  
Thanks very much!!

However, I tried this with today's latest version, 2.7.91, and in that 
version, my CAT port for the mcHF does not work.  I backed down to 
2.7.88, and it again works fine, so I'll stay there for now.  Just to 
double check, I changed back to 2.7.91, and it still doesn't work.  So 
back to 2.7.88, and all is well again.

I was going to report this on github, but saw a message to place "test 
reports" elsewhere.  When I looked there, all I saw was in German, 
which, unfortunately, I don't read.  So this report will have to do.

73,
Randy, KS4L

db4ple, 2018-02-05:
Hi Randy,
please help us by testing .89 and .90 as well and tell us the version which broke CAT for you. Please use a separate (new) issue for that. It is most likely absolutely unrelated to the topic discussed here AND this issue is closed anyway.

TNX,
Danilo

df8oe, 2018-02-05:
Hi Randy,

please feel free to post there in English. It is a multilanguage community - many developers come from Germany. So English AND German are the languages there. Bug reports indeed must be placed here on GitHub.

df8oe, 2018-02-05:
Regarding CAT: I just tested from 88 to 91 but all versions do work flawlessly using same settings on software. Cannot confirm not working CAT.

KS4L, 2018-02-05:
Danilo, 

Is it necessary to reset or adjust any Menu settings related to CAT when switching between these versions? I did not reset or adjust an settings when I checked this yesterday.

73,
Randy, KS4L

ve3ids, 2018-02-05:
2.7.89 and .91 work fine with cat control on my version 6 mcHF. I didn't
try .90
73 Don ve3ids

>
> Regarding CAT: I just tested from 88 to 91 but all versions do work
flawlessly using same settings on software. Cannot confirm not working CAT.

—
<https://github.com/df8oe/UHSDR/issues/1011#issuecomment-363076252>, or mute
the thread

df8oe, 2018-02-05:
It is not neccessary to change anything - everything works out-of-the-box as before.

KS4L, 2018-02-06:
Further info on my problem with CAT in my mcHF 0.6. I am testing this with
the latest WSJT-X (1.8.0) running on Windows 10 with the latest updates.  I
loaded and tested all the FW versions from 2.7.88 to 2.7.94 (today's
latest), running BL 3.4.6.  My mcHF CAT works correctly using WSJT-X
(1.8.0) on all the tested FW versions except 2.7.91, 2.7.92, and 2.7.93
where WSJT-X fails to find the COM port for the mcHF. Even when WSJT-X
fails to find the COM port, Windows Device Manager still correctly shows
the COM port (COM 5 in my case), as well as the mcHF USB audio ports.

This could be a problem with WSJT-X, I suppose, but this version of WSJT-X
has been out for several weeks and I haven't seen any other complaints
about this.  It also works fine with my Elecraft K3.

However, if no one else can reproduce this, I'll write it off, since the
latest mcHF version (2.7.94) works fine for me.

And in any case, all of the FW versions 2.7.88 - 2.7.94 work well with
straight key input for CW, which was what this thread is about anyway.

73,
Randy, KS4L

ve3ids, 2018-02-06:
Randy,

I didn't test mine with WSJT-X, I was using DXSUITE Commander to test.
Sorry, I didn't think to try WSJT, I will try that when I get a chance.
Which version are you running, mine is probably 3 months old.

73 Don VE3IDS

KS4L, 2018-02-07:
I’m running WSJT-X 1.8.0.

> On Feb 6, 2018, at 3:25 PM, ve3ids <notifications@github.com> wrote:
> 
> Randy,
> 
> I didn't test mine with WSJT-X, I was using DXSUITE Commander to test.
> Sorry, I didn't think to try WSJT, I will try that when I get a chance.
> Which version are you running, mine is probably 3 months old.
> 
> 73 Don VE3IDS
> 
> On Tue, Feb 6, 2018 at 4:19 PM, KS4L <notifications@github.com> wrote:
> 
> > Further info on my problem with CAT in my mcHF 0.6. I am testing this with
> > the latest WSJT-X (1.8.0) running on Windows 10 with the latest updates. I
> > loaded and tested all the FW versions from 2.7.88 to 2.7.94 (today's
> > latest), running BL 3.4.6. My mcHF CAT works correctly using WSJT-X
> > (1.8.0) on all the tested FW versions except 2.7.91, 2.7.92, and 2.7.93
> > where WSJT-X fails to find the COM port for the mcHF. Even when WSJT-X
> > fails to find the COM port, Windows Device Manager still correctly shows
> > the COM port (COM 5 in my case), as well as the mcHF USB audio ports.
> >
> > This could be a problem with WSJT-X, I suppose, but this version of WSJT-X
> > has been out for several weeks and I haven't seen any other complaints
> > about this. It also works fine with my Elecraft K3.
> >
> > However, if no one else can reproduce this, I'll write it off, since the
> > latest mcHF version (2.7.94) works fine for me.
> >
> > And in any case, all of the FW versions 2.7.88 - 2.7.94 work well with
> > straight key input for CW, which was what this thread is about anyway.
> >
> > 73,
> > Randy, KS4L

ve3ids, 2018-02-08:
Randy

I just tried mine and wsjt-x works normally with 1.8.0 and 2.7.91  but I am
running Vista though so that may be the difference.

73 Don ve3ids

I’m running WSJT-X 1.8.0.

>
> Randy,
>
> I didn't test mine with WSJT-X, I was using DXSUITE Commander to test.
> Sorry, I didn't think to try WSJT, I will try that when I get a chance.
> Which version are you running, mine is probably 3 months old.
>
> 73 Don VE3IDS
>
>
> > Further info on my problem with CAT in my mcHF 0.6. I am testing this
with
> > the latest WSJT-X (1.8.0) running on Windows 10 with the latest
updates. I
> loaded and tested all the FW versions from 2.7.88 to 2.7.94 (today's
> latest), running BL 3.4.6. My mcHF CAT works correctly using WSJT-X
> (1.8.0) on all the tested FW versions except 2.7.91, 2.7.92, [...]
WSJT-X
> has been out for several weeks and I haven't seen any other complaints
> about this. It also works fine with my Elecraft K3.
>
> However, if no one else can reproduce this, I'll write it off, since the [...]

—
<https://github.com/df8oe/UHSDR/issues/1011#issuecomment-363747741>, or mute
the thread