df8oe/UHSDR #1013: Volt Meter No working

g7jiq, 2017-08-22:


g7jiq, 2017-08-22:
Has anyone noticed that the voltmeter isn't working as it should.
I have just noticed after changing power supllies that its stuck about 12.3 volts, this does drop to 11..2 when I drop the supply voltage to 7 volts.
I am using the latest (D2.5.60) software.

df8oe, 2017-08-22:
No. I do not have noticed because no of my mcHFs does show this behaviour. All shows voltage as expected.

So I think you do have a hardware problem in voltage divider stage or connection between this stage and STM32...

s53dz, 2017-08-23:
Hi,

It works well on my mcHF using FW version D2.5.60.

73 Bojan

ve3ids, 2017-08-24:
My voltmeter was at about 12.3v as well but I never really payed attention to it. I went into the calibration menu and it adjusted up to match the 13.74 of the measured supply just fine and seems to be accurate now. Have you calibrated yours yet?
73 Don ve3ids

df8oe, 2017-08-24:
I assume this is no firmware issue but a hardware issue... So I think it can be closed...

73
Andreas

g7jiq, 2017-08-24:
Hi All
Thanks for your help.
I have calibrated it to match my supply of 12v, (had done this when first built) and if I drop the supply from 12.0v to 9.0v it shows 11.95v and if I up the voltage to 13v  I get 12.7v, and down to 6.0v it reads 11.51v. 
I seem to remember when it was first built it worked ok, 
Possibly something has died, 
Can anyone suggest which section detects voltage.

Thanks
Dave

df8oe, 2017-08-25:
RF board R13, R14, C31 and all their connections (to V-input, GND and STM32F4)

vy 73
Andreas

g7jiq, 2017-08-25:
Thanks.
Will have a look and see if i can find anything out of place.

closed 2017-08-25 by g7jiq