df8oe/UHSDR #1014: LED dimmer

AngelSerra, 2017-08-23:
When you press the power button reduces the brightness of the screen, could it possibly be the same with the RX and TX LEDs?

df8oe, 2017-08-23:
If you want to do so you have to drive the LEDs using PWM instead of DC voltage. Of course this is possible, but there are many disadvantages so it is not planned to do that:

1)You have to generate PWM using CPU time which causes more load (we do need CPU power for much more important tasks)
2)Maybe due to new square wave outputs of MCU you will have more noise at the lower bands
3)You will have new interrupts which probably may impact functionality of mcHF.

DG9BFC, 2017-09-05:
if we would have free gpio pins it should be possible without pwm (but we have not cause chris used them all) ... one pin would drive the led direct (bright) and another pin drives it via a resistor (dimmed)

(maybe an idea for the i40?!? ... but in my view that dimming of led is not needed)

closed 2017-10-01 by df8oe