df8oe/UHSDR #1020: RX and TX marker lines

s53dz, 2017-08-27:
Now, when the second marker line is so well introduced to the RTTY mode, why not extend the same principle also to other modes by adding an RX vertical marker line beside the TX one? I think this may help to have a clear overview when using RIT and/or SPLIT functions.

73 Bojan

DG9BFC, 2017-09-05:
these marker lines are included now ... so ... in my view the issue can be closed or ... oh wait ... you mean you wanna have anther line  if rit is not zero???

s53dz, 2017-09-05:
Sigi,
Until the end of this week unfortunately I can not test this markers but the only reason for having the RX marker is when it is different from the existing TX one, aye? (CW, SSB modes)

73 Bojan

s53dz, 2017-09-18:
Sigi,
There is no RX marker line added but I can live also with a TX vertical marker line and an RX filter horizontal line as it is now.

73 Bojan

closed 2017-09-19 by s53dz