df8oe/UHSDR #1028: FW size - question

s53dz, 2017-09-01:
Now (D2.5.68) with STM32F4xx we are at the .bin/.dfu file size of 443kB. Already approaching the MCUs with lower flash-size limit.
I wonder what will the way of UHSDR FW in the future be? Branching for:

a) F4 512kB flash
b) F4 1MB flash
c) F7

and leaving a F4 512kB branch as is?

I suppose at the moment there are three groups of radios out there with 512k, moderate with 1M, no with 2M and a few with F7. I belong to the first two groups.

73 Bojan

db4ple, 2017-09-01:
Hi Bojan,
that is an interesting question. Well, I think we will manage to have a stable core configuration which will remain below 512k flash and 192k RAM usage. For instance if we leave out FreeDV support, we’ll loose a lot of weight both in terms of RAM and flash.

We do not plan to create a special branch for that configuration, it is just a configuration, like the OVI40 F7 is one. We are gradually preparing the software to better cope with extensions such as different display or key layouts.  In addition we introduced the concept of "USE_" defines, which control the capabilities included in a build. E.g. USE_FREEDV is one of these, USE_SNAP another.  There are a few more of these.
In combination with the hardware definition, this will give us enough flexibility without too much performance and resource impact. 
In consequence this means for new features, which require more RAM and flash, implementation should be done in a way that we can turn it off without impacting the base functionality. 

We may come to that point later, at which we will drop support for the F4 / 512k, but then this would mean an end of development for this configuration (No support, no bugfixes) unless someone volunteers to do so. And honestly, it would be better to resolder a larger MCU at this point, that is 30min work at most. 
Being able to enable only a subset of all available features is a generally relevant concept, since although the newer MCU have more resources, normal RAM is still at a premium.

BTW, to be honest, it is one of my personal interests in that project to build a true platform instead of just advancing with new features by switching to more powerful hardware and leaving the "normal" user behind. Kind of QRP spirit for trx development, you could say. That this has a price I know only too well.

s53dz, 2017-09-01:
I like the idea of using "defines" for the selection of wanted and/or possible functionality. I am also convinced that each one who builds today uses a 1MB flash size version of MCU.

df8oe, 2017-09-01:
That was one point where Chris agreed to my opinion very fast. I think since 1.5 years all kits and preassembled boards which he has sold use 1MB chip...

s53dz, 2017-09-04:
Just making a note for now about the information given here and a possible issue in the future.

closed 2017-09-04 by s53dz

s53dz, 2017-09-26:
Well, I am now on 2 x STM32F429VGT6, since I have changed the old 407VE. So I am fine for some time.