df8oe/UHSDR #1035: Issue with FW 2.5.73 Bug?

wk8s, 2017-09-06:
When I key the mcHF under 2.5.73 the TX LED stays lit and radio cannot be powered off until I unplug it from power source. Reloading FW 2.5.72 restores functionality to normal. Not sure if anyone else sees this behavior.

df8oe, 2017-09-06:
I can confirm that here is something faulty. I must not remove power (it helps to press TUNE twice) but of course that is not ok.... Thank you - we will take a look and fix it.

db4ple, 2017-09-06:
Please state the mode (CW, USB etc.) in which the problem appears. 

TU
Danilo

db4ple, 2017-09-06:
And if it happens at the first TX after power up or at the second TX (or third ... ). 

73
Danilo

wk8s, 2017-09-06:
It happens in CW or Tune every time. I did not test other modes. Happens on any band. Doesn't actually stay in transmit mode with RF output but TX led stays on and receive is muted. 

Pete Meier
pmeier@me.com
Amateur Radio Callsign: WK8S
|| I never lose. If I don't win, I learn. - unknown author ||

db4ple, 2017-09-06:
Thanks! Btw, in such a case, please test also other modes, if these modes exhibit the same problem. 
I test with only my commit (#1033) and all modes seem to be working ok, so I think this is related to the #1034 commit which in fact changes the CW state machine quite a bit. 

@phaethon: Please have a look.

df8oe, 2017-09-06:
I can confirm that it only happens in CW. Sometimes it works a few signs, then TX LED keeps red, no CW sidetone. You can return to RX by pressing TUNE twice. I cannot reproduce with TUNE (regardless of mode). So I think it is a CW issue.

phaethon, 2017-09-07:
Yesterday evening I was working with the latest code and didn't notice any issues. Can you be more specific when/how it happens and how to repeat it?
Please, comment the key mode (iambic, straight, ...), WPM and CW txrx delay settings (and if it is 0ms, then whether the issue repeats if you change it to something higher).

df8oe, 2017-09-07:
Yesterday I was able to reproduce it 100% after a few seconds. After I built new fw 2.5.74 it was already there after a few seconds (keyer mode ULTIMATIC, 15WPM, TX/RX Delay 10ms). But after this I was able to key several minutes without the issue - cannot reproduce at the moment. But I am sure: it is already there, but it is intermittent. Maybe a race condition. Will report if I do know more.

wk8s, 2017-09-07:
This morning (in USA) I loaded 2.5.74 and can verify it happens in CW mode every time - 18WPM, Iambic B, 0 delay - TX goes on, RX muted - RF output only on key-down. Two presses of TUNE and TX light goes off and receive goes normal. Does not happen in other mode USB, LSB or RTTY. Can confirm it does NOT happen in TUNE.  I believe it's related to the TX-RX DELAY - I tested several times at 0, 10, 20 and 30ms delays. IT DOES NOT HAPPEN AT 10, 20 OR 30ms. It does happen a 0ms EVERY TIME. Hope this helps.

phaethon, 2017-09-07:
Yes, this helps. I will check cw state transition. Possibly, putting infinitesmal delay when txrx setting is 0 would be a workaround, but I will get to the radio to test not earlier than late in the evening.

db4ple, 2017-09-07:
Without looking to deeply in the code changes but with my background in the original code and given the error conditions from @wk8s  : It could be related to the cw internal timers. I had a similar problem, where the timers would go negative (which since these are unsigned int means they will be become large positive numbers) and then it takes a loooong timers before the times turn off (once it passes zero).

Easy to find, if you are using an ST-Link and you can recreate the error condition, then a quick look at the relevant timer values will tell you if my hunch was right.

df8oe, 2017-09-07:
It is a timing- or race-condition. Using new built 2.5.75 I cannot use CW at all. Issue appears instantly when you key the rig. The hint reagaring CW timers sounds matching.

wk8s, 2017-09-07:
CW works properly at 10ms delay for me with 2.5.75 but NOT at 0ms where it is instant as before when key rig.

df8oe, 2017-09-07:
I am unable to transmit one single sign @10ms. TX stays steady at first keying.

df8oe, 2017-09-08:
Same in 2.5.76. I cannot key one single character. If I press paddle (regardless if dit or dah) mostly nothing happens (like malfunctioning contacts). If there is a reaction: mcHF goes to TX (red LED on), but no RF output and no sidetone. Now it is stuck in this state. You can leave it by pressing TUNE button twice.

df8oe, 2017-09-08:
Additional information:
Straight key does not work, too. Same behaviour as IAMBIC / ULTIMATIC.

df8oe, 2017-09-08:
I will test and report!

df8oe, 2017-09-08:
Result:
I cannot key one single sign. There is no difference in behaviour of D2.5.76 and the version on your GitHub.
Regardless if CW RX/TX-Delay is set to "0" or to any other value: Mostly nothing happnes when I press one paddle. Sometimes there  is one short siedtone and mcHF remains in TX mode without sidetone, absolutely rarely ( I was successful only one try out of hundreds) I could key one sign correctly.

phaethon, 2017-09-08:
@df8oe 
Code on my version fixes only the case when it works with >0 txrx delay and doesn't with 0 delay. Like for @wk8s 
Give me time until the end of the weekend, and if I will not find the bug I will revert the changes in cw_gen and create a separate branch until making it stable. 
Do you have any relevant hardware differences? I am testing on M0NKA v5 set. And it works well.

df8oe, 2017-09-08:
I do not have special hardware or special modifications. I just draw back to 2.5.72 and it is working flawlessly. All other versions after that (include 2.5.77) do not work. Not able to key. Not using paddles, not using straight key. Sometimes no reaction, sometimes only TX LED but no sidetone (and can only go back pressing TUNE twice) and sometimes (very, very rare TX AND sidetone AND RF output. But maximum 1 character - then it hangs LED on, no sidetone, no RF output.

db4ple, 2017-09-08:
Hi @phaethon  or @df8oe , since this bug has a longer than wanted lifetime, please change the title to indicate the nature of the issue. The version number should remain visible too since it started with 2.5.73

TU
Danilo

s53dz, 2017-09-09:
D2.5.80 quick report:

It seems the CW keying works now, only the red TX LED is now not responding always on TX.

73 Bojan

s53dz, 2017-09-09:
Correction:
I don't know if this is the same issue, however after a while on receive with CW decoder switched on, mcHF hangs up.

73 Bojan

df8oe, 2017-09-09:
I am absoultely astonished... Since 2.5.80 I can key using TX->RX Delay != 0 wihout problems. Using delay == 0 it hangs.

@Bojan:
CW Decoding is a new pre-alpha feature (unstable). Thank you for your report - that is not the same bug and there is no need to create a new issue report we are already working on that.

db4ple, 2017-09-09:
Maybe put "CW TX bug" in the title to be more precise

closed 2017-09-10 by df8oe

wk8s, 2017-09-10:
> Closed #1035 via da22fba.

Thanks for all the work to resolve this issue. 

@WK8S

commit da22fba8 (2017-09-10): Merge pull request #1050 from phaethon/active-devel

Reverting some of the earlier cw keyer related changes to fix #1035