df8oe/UHSDR #1041: Support for external VFO

sm0tsc, 2017-09-08:
Can we have support for external VFO via USB similar to https://store.griffintechnology.com/us/catalogsearch/result/?q=powermate

//SM0TSC

df8oe, 2017-09-09:
I don't own such a device (and I by myself will not buy one). But if you send me one so that I can investigate as what it identifies itself I think we can do that.
Additionally you can help us by connecting it to a Linux machine and post the result of lsusb -vv.

But pay attention:
USB host at STM32 is a very simple one. You cannot connect an USB hub to add more than one device. If you use such a plug this will prevent you from using a keyboard for RTTY and other following digital modes! Why having one input device twice and another then is impossible to connect?

73
Andreas

sm0tsc, 2017-09-09:
I don´t currently have one. But I will try to get hold of some.. It was
just an idea popped up in my head :-)

73
Johan - SM0TSC

closed 2018-01-30 by df8oe