df8oe/UHSDR #1044: carrier marker line in mode ssb is not allways drawn correct

DG9BFC, 2017-09-09:
title says all but i have to explain what i mean ....
when i switch on the mchf in mode lsb the marker has full (normal) height) ... when i change mode (around the clock with pushing mode button a few times till i am back to lsb) then the marker of the carrier point only has about one box height .... all other marker lines are ok but in ssb it is not

df8oe, 2017-09-09:
I just tried to reproduce - no success... Marker line is drawn from bottom to top with waterfall speed in every mode...

db4ple, 2017-09-09:
Probably Sigi meant Scope Mode. Important to specify which mode the spectrum is in.

g7jiq, 2017-09-09:
I have had a play with the software and I get the same problem,   (only in scope mode)
If you cycle thought the modes the tuning line disapears in the noise on LSB and USB then returning on the next mode.
From ver D2.5.72 to present ver this happens. (have tried them all)
From Ver D2.5.60 to D2.6.71 the left tuning line on DI-U disapears to the first box as before, and returns on next mode ( LSB.)
On ether case if the menu is selected and canceled the line retuns as expected at the screen is refreshed.
One other interesting thing is if left undisturbed the line slowly grows back...... 25 mins gave me 2 out of 4 boxes.
Hope this is of help..

df8oe, 2017-09-09:
So it is - only in scope mode..... I haven't used scope mode since over 1 year :)

g7jiq, 2017-09-09:
Yes its only in scope mode, 
I only use scope mode, as due to a very high noise level the waterfall is of no use, even after adustments.

DG9BFC, 2017-09-09:
oh yes i should have mentioned that it is in scope mode ... a strong carrier draws it back (cw carrier as an example)

g7jiq, 2017-09-10:
I have found after installing the latest update D2.5.88, if you select RTTY not FreeDV the tuning line remains throught out the mode cycling.
This is good enough for me......

DG9BFC, 2017-09-17:
seems to be solved so can be closed (for now)

closed 2017-09-17 by DG9BFC