df8oe/UHSDR #1049: E2 second function

s53dz, 2017-09-09:
When cycling thru modes (using S16) and when passing the RTTY mode the M2 (S3) switch switches to its second function (RTTY SHIFT setting) and stays there.

(FW D2.5.82, but probably also before that.)

73 Bojan

db4ple, 2017-09-09:
Yes. This is a  "work in progress" area since the old code lacks a little in that department (flexible switching encoder layouts between modes was all handcrafted code. This gets with each mode more difficult to maintain and change).

s53dz, 2017-09-10:
Danilo,

I understand it is not easy to maintain all settings between switching modes, that is why I started this issue in an easy way: if it is possible just to remove pushing the second function when in RTTY mode it would release this issue then.

What happens is this: You switch modes across and end-up in SSB and try to change AGC / RF gain, but you are changing the RTTY shift. To get back to AGC you have to roll thru a few of E2 functions.

73 Bojan

closed 2017-10-09 by s53dz