df8oe/UHSDR #1057: CW low wpm hangs

I2KBD, 2017-09-10:
Using CW decode fw 2.5.87 is wpm drops to 3-4 even is LED is flashing following CW signal correctly, speed never tracks again signal and fail to decode.

I2KBD, 2017-09-11:
Happened again today with fw 2.5.92
wpm indication start to decrease (even with a CW signal) and when it reaches 4 wpm there is no way to lock any signal anymore unless I turn radio off and on again.
Alberto I2KBD

s53dz, 2017-09-12:
I suppose we do not need the CW decoding of signals with say WPM < 15. Where I mean an element duration corresponding to WPM15 and not the actual words count per minute. The later could be less. I feel lower speed CW signals are not easily decoded by listening. For QRSS reception, as an example, there are programs to do that better.

Perhaps this could help to limit the hanging of the decoder.

73 Bojan

df8oe, 2017-09-12:
I think disabling decode at < 15WPM is not neccessary. We played a little bit wit setting "Signal Threshold" and lowering it a bit does solve the problem. Additionally we will implement a possibility to "reset" decoder so that it starts at ~20WPM again.

73
Andreas

I2KBD, 2017-09-12:
Hi Andreas,

I agree with you. We have several beacons working at 12-14 wpm that is interesting to decode.

I don't know why, when speed drops below 5-6, it cannot recover when receiving a faster signal.

Many, many thanks anyway for your great effort!

 

Alberto I2KBD

A: df8oe/UHSDR
Cc: I2KBD; Author

 

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

DG9BFC, 2017-09-12:
hello alberto
you can also try to play around with decoder settings
one major point is the blocksize ... that highly influences the decoding speed and so the wpm result as far as i can test here
the programmer maybe could a bit explain what settings in the decoder influence what ... easier as only try and error

DD4WH, 2017-09-12:
Hi Sigi,
the point is that a CW decoder is quite a complex system, so I cannot tell you exactly how the decoder settings influence decoding quality. That is why we are very interested in test results with different settings. Unfortunately, we have not yet received any information on that from people who have tested the decoder . . .

db4ple, 2017-09-13:
BTW, my fix is not working as good as I found in my first tests. Further testing reveal it still looks in the lower range. We need to look more closely into this and understand the algorithm better. Please do not report "it still doesn't work". We know that. We would rather want to learn when it works and when it stops working.

wk8s, 2017-09-13:
Not sure this a bug or when it started but noticed the Step Size Marker gets overwritten and disappears when changing frequency and does not re-appear unless you restart or turn off/on in the Menu. Currently have fw .94 installed

db4ple, 2017-09-17:
Please close the issue, it seems to be gone.

wk8s, 2017-09-17:
Good Morning,

I did not open the issue #1057. I did make a comment on it. I am unclear on what I should do. I can edit my comment or make another but it doesn’t seem appropriate that I close the issue which I did not start. Please advise and I will follow up.

Pete WK8S

> 
> Please close the issue, it seems to be gone.

Pete Meier
pmeier at me.com
Amateur Radio Callsign: WK8S
|| I never lose. If I don't win, I learn. - unknown author ||

db4ple, 2017-09-17:
Hi Peter, 
this message was intended for the original issue poster ( @I2KBD ) or the repo owner ( @df8oe  ). You and I can't close this issue due to lack of permission, but we still get all the mails related to it.

73
Danilo

wk8s, 2017-09-17:
Thanks that what I thought, although the comment I made about the step marker has been fixed. I will edit my comment stating that.

73, 

Pete

> 
> Hi Peter,
> this message was intended for the original issue poster ( @I2KBD <https://github.com/i2kbd> ) or the repo owner ( @df8oe <https://github.com/df8oe> ). You and I can't close this issue due to lack of permission, but we still get all the mails related to it.
> 
> 73
> Danilo

Pete WK8S

	
pmeier at me.com

closed 2017-09-17 by df8oe