df8oe/UHSDR #1059: receive audio mute

sv1it, 2017-09-11:
Hi,

I experience a receive audio mute from time to time, which is cured only by restarting the mchf. Any ideas what the problem might be?

73

de Kostas sv1it

df8oe, 2017-09-11:
Taht can be related to many different reasons. It is not a firmware issue: otherwise much more OMs must have reported this.

So I can only guess:
1) too high I2C-speeds selecte (try with 100 or 120KHz)
2) spikes on your voltages 5V / 3.3V?
3) bad solderings around audio codec?
4) defective LM380?

Can you please test if USB audio is present if your speaker is muted? Or line out audio?

73
Andreas

sv1it, 2017-09-11:
Hallo Andreas,

 

Danke für die schnelle Antwort. In welchem Menü kann man das I2C speed einstellen?

 

73

 

Kostas SV1IT

db4ple, 2017-09-11:
Hi,
thanks for reporting through github. You gave additional information in the Yahoo NG, please add this. Not all developers read the Yahoo NG but all read Github issues.

We need some more information:
You should tell us which firmware version you are using (shown at startup, also which hardware version) and typically a screenshot will help to see the most basic settings you have active.

Based on the information in the Yahoo NG, I don't think it is a hardware issue (since you probably never changed the I2C speed) and it seems to happen in the middle of a receive OR does it happen when switching back from TX to RX?

If it happens in the middle of a receive then it is most likely an AGC related issue, since this is the only part which will change the volume of the received signal in the middle of a receive.
 
During switching we have some muting code active, but this only applies if sound goes missing after returning from TX.

sv1it, 2017-09-11:
Hi, thanks for reply.

 

Firmware is 2.5.91, installed yesterday. Hardware is the rs-918 (Chinese clone). Most of the time the problem occurs after going from transmit to receive, although I think it once happened during receive (I am not absolutely certain about this though). It is definitely happening during switching from transmit back to receive, but not always, maybe once every 5 transmission on average (this is a guess). The scope continues working as if reception is OK, even with audio muted.  

 

73

 

De sv1it

db4ple, 2017-09-11:
Hi,
please DO NOT INCLUDE the original mail text in the replies. This clutters the communication incredible in the github interface. Just delete all old stuff before sending. Thank you!

Send in a screenshot of your normal settings. In Github Web Interface you can simply drag picture into the comment field.  
As I said, this sound like a AGC related issue. It is also important to report the mode and the settings related to AGC and RF Gain in the menus. I know these are quite a lot but without that we may not be able to reproduce the problem.

mcafzap, 2017-09-12:
Two points:
1. No everyone will report such intermittent faults especially when on/off is such an easy work around.
2. A transceiver is full of electromagnetic radiation and hence _bound_ to cause interference within its own environs. Hence a sudden change of power from Rx->Tx or Tx->Rx might leave those devices which are serially controlled, in an unknown state. IMHO re-initialisation of such devices should therefore be carried out whenever possible... .

df8oe, 2017-09-12:
I do not agree. Everything is reported - may it be as small as you can think. we get issue reports here on GitHub, at Yhoo NG and in our German discussion group. Noone has reported that the last months.

If hundreds of mcHFs do run without this double/triple initialization I think we must NOT add it. Maybe quality of Chinese clones is too bad? We should investigate the reason for that behaviour...

genesilvernail, 2017-09-15:
I have been living with similar issue on a 0.6 mcHF build, not PRC. This is on a unit with no case or inter board shield which I attribute, maybe, pertinent. It's AGC related since , at times, the depth and duration of audio loss varies. At times the audio will slowly return, 2-5 seconds, the balance of times requires power cycling to restore. All this on return to receive from transmitting (ssb). The SW is 2 months old and will verify exact tomorrow.

Listening with keen interest....thanks

DD4WH, 2017-09-15:
Just my two cents:
- everyone using chinese clones should be aware that buying an illegal clone often leads to issues in the functioning of the hardware. I am not motivated to solve such problems.
- it seems the problem is related to the AGC settings (as was already mentioned). I strongly recommend to use the WDSP AGC and adjust it according to the instructions in the WIKI. Could well be that the problem is already solved then.

https://github.com/df8oe/UHSDR/wiki/Automatic-Gain-Control-(AGC)

73 Frank

sv1it, 2017-09-15:
Hi Frank,

 

Thanks for replying to my issue. First of all a “philosophical” point: are there intellectual rights patented related to the mchf? If not, why should one condemn a copy of it and why should it be termed ‘illegal”? This is obviously a questions for specialist lawyers, which I am not, however IMHO it is an exaggeration to call the Chinese builds “illegal”.

 

Second point: I thank you for the technical part of your answer. I will play around with it and report my findings.

 

Best 73

 

Kostas SV1IT

df8oe, 2017-09-15:
Hi Kostas,

the chinese clones *are illegal* because their existence is a license break of mcHF hardware. It is licensed under creative commons BY-NC-SA - please take a look at Chris' homepage. Not to know this doeas not legalize anything.

+++++++++++++++++++++++

First of all we are giving support for Open Source UHSDR firmware. If the issue can/may be software related, their will be support. If we cannot see any software related problem we *may* have some hints for self built mcHFs. But if we know someone has got one of the chinese clones please understand that we will not give any support. First of all it would be a good idea to open the case and investigate which "recommended modifications" are missing. That is a point everyone who builds or has built a mcHF has struggled around and you cannot get rid of it by buying clones.

vy 73
Andreas

DD4WH, 2017-09-15:
Hi Costas, 
intellectual property does not need to be patented in order to make those clones illegal. Thanks Andreas, for clarifying.

I am doing the work on the UHSDR firmware in my free time, so it is my decision how to spend this time. 

Just accept that I personally will not give help/advice for illegal clone users.

73 de Frank

sv1it, 2017-09-15:
Hi,

 

I believe the concept behind creative commons is that ideas should and cannot be patented by anybody. Hence the concept of braking an open source product is – I believe – pure nonsense.

 

Best 73

 

Kostas SV1IT

DD4WH, 2017-09-15:
Hi Costas,
your belief is absolutely wrong. Please inform yourself about the idea of open source. And BTW, creative commons has many different license options.

And my statement about illegal clone users still holds.

My suggestion would also be to stick to the title of the issue again (Software issues of receive audio mute).

73 de Frank

df8oe, 2017-09-15:
Hi Kostas,

I agree that I have had problems with the license type Chris has used for mcHF from the beginning on.

And I also agree that it is not "real Open Source".

I am happy that UHSDR is licensed under GPL v3 since ~ half a year and OVI40 would be "much more open". It will be released under "CERN OHL version 1.2".

But we cann not change anything for mcHF because it is created by Chris M0NKA.

vy 73
Andreas

sv1it, 2017-09-16:
Hi Frank,

 

First of all let me tell you that I have no connection to the Chinese manufacturer, I did not know anything about the mchf project and its existence until a few weeks ago. I have built myself over the last 10 years a number of sdr equipment and enjoy it very much. I heard about the rs918 from a fellow radio amateur, checked the internet, found that I could buy it online for 350US$ and ordered one, since it sounded very good value for money. Later, after I had ordered the rs918 I searched further and found about the mchf project and the controversy regarding licencing. I wrote the email you just responded to on a “theoretical” or “philosophical” basis. In principle I do not believe in intellectual rights generally, because I think humanity has progressed in science, technology and “ideas” through the continuous, common effort of billions of human beings through the centuries. No one can claim that he came up with any idea on his own. Everything we know about or build is the result of that common effort through the millennia. Therefore for somebody to claim rights on anything is really not based on logic but rather on personal “standards”.

 

Now of course you have every right to answer/help to anybody you believe has “trespassed” some limits you believe should not. Thanks anyway.

 

Very 73

 

De SV1IT, Kostas

sv1it, 2017-09-16:
Hi Andreas,

 

After all those discussions on patents, open source etc. I tried to concentrate on the receiver muting issue. Today I downloaded the latest firmware (2.5.101) from the UHSDR, made the AGC adjustments recommended and BINGO everything seems to be working fine. Thank you and thanks to Frank as well for the support and the “lively” discussion!

 

73

 

Kostas SV1IT

db4ple, 2017-09-17:
Hi Kostas,
please DO NOT INCLUDE THE HISTORY OF THE ISSUE IN YOUR REPLIES. Have a look at the web interface and see how difficult this makes to read the comments. You can also edit your comments and remove the duplicate content! Thank you. Once done, I will also remove this comment...

sv1it, 2017-09-17:
Ok, sorry. I will take care next time.

phaethon, 2017-10-04:
Is this issue still open?

df8oe, 2017-10-21:
I close this because there are no facts about the issue told. Maybe it will come up again with more facts.

closed 2017-10-21 by df8oe