df8oe/UHSDR #1063: After CAT band change, BAND +/- goes to wrong band

KS4L, 2017-09-11:
After doing a band change through WSJTX (CAT), BAND +/- does not go to the correct band. Instead it acts as though the CAT band change had never occurred.  For example:
- Starting on 20m with VFO A
- Change to 17m through WSJTX (CAT)
- Press BAND- sets mcHF to 30m, not back to 20m
Also, the not-active VFO (B) is not changed by the CAT command as it is with BAND +/-.
Also, the sequence of bands for BAND +/- remains wrong until a power cycle.

73,
Randy, KS4L

db4ple, 2017-09-12:
Hi,

this is normal. You can achieve the same by dialing a different frequency by hand. Each band memory can be dialed to any frequency. The label (right , below meters) will show you the Band Memory Name.  For CAT you can set in the configuration menu "CAT in a sandbox" to "On". This will prevent the band memory from being overwritten by CAT commands. You'll see "CAT" in the label field then.
Switching bands manually makes the transceiver  "forget" the CAT frequency change.

73
Danilo

KS4L, 2017-09-13:
Thanks! CAT in a sandbox works better.

db4ple, 2017-09-13:
Please close the issue then. Thanks! Danilo

closed 2017-09-13 by KS4L