df8oe/UHSDR #1069: Waterfall direction

Jmelz, 2017-09-14:
Has there been any thought to reversing the waterfall direction? Most other applications use a waterfall that scrolls down (like an actual waterfall).

If the issue is that the consensus is to keep the frequency counters next to the spot where the waterfall 'starts' from, why not move those to the top as well then?

@ #

df8oe, 2017-09-17:
...because that is much work and it is working "as it is" too, or not ;)

Jmelz, 2017-09-17:
Thanks for the reply. It was just unusual to change in my head a waterfall from falling down to falling up :)

Jmelz, 2017-09-24:
I saw this today:

https://www.youtube.com/watch?v=HzEP_i8pxCs

Looks like someone was able to reverse the direction and then completely changed the layout as well to suit the larger screen they used.

df8oe, 2017-09-24:
Yes. But that is not mcHF - it is Raspberry Pi powered and completely other hardware.

Jmelz, 2017-09-24:
Really? I read this in the description: " Ui board same as for original mcHF, mcu STM32F405, except one additional adapter board between LCD and UI board "

df8oe, 2017-09-24:
Really! STM32F4 does not have enough RAM to hold waterfall AND spectrum data of larger LCD.

But: not a problem to reverse it. If you do have much time and the knowledge: you are invited to do it.

I will not waste my time for things which are not functional issues but only optical ones. It is a community project - just start over...

db4ple, 2017-09-24:
Hi Andreas,
I don't think, you are right. The text says "3.5 RPi Display" not RPi and the youtube user says modified screen routines (much faster) but using the 405. So it is the genuine thing. Hopefully the youtube user will release the source code and will not just sit on it so that we can see how he did this trick. 
There is enough RAM if you leave out FreeDV I would say. Maybe even without this.

73
Danilo

Jmelz, 2017-09-24:
Yeah I figured as much with displaying both. I'm a complete novice with code - I typically beta test for several projects, and got into the mchf for the low cost and speedy updates you provide. I am quite incapable of making any useful changes. 

No worries on the topic - I just wanted to say that it looks like it could be done. I figured many people would be interested in having this display the waterfall correctly, but if there is not interest, and no one is pushing (or even helping with the code), then I can see it is not worth the time right now.

db4ple, 2017-09-24:
BTW,
reversing the direction is actually quite easy. But I, like Andreas, do not see the point of this change at the moment. So this is left as an exercise to our readers.

df8oe, 2017-11-04:
included in #1149 and therefor closed.

closed 2017-11-04 by df8oe