df8oe/UHSDR #1071: Beep volume not changed when using headphones

Jmelz, 2017-09-15:
Beep volume at zero does not mute beep volume when using headphones.

frazierjason, 2018-02-11:
This bug applies to speaker out mode as well -- not just headphones.  You just can't hear the zero-level button beep from the speaker unless you turn up the audio gain a lot.  It just so happens that the beeps come across into headphones (especially big can headphones) much more loudly than from the unit's speaker.

DG9BFC, 2018-02-14:
also in version 2.9.3 same issue ... setting level to zero does not mute the keybeep (and in my view ther level could be much lower in lower settings) ... switch OFF does work but level zero does not mute completely
lower the keybeep level to half or a quarter of what it is now and it is still loud enough ... level zero could be same as switching it completely off  (it should show OFF instead of "zero") and the on off setting could be deleted from menue ... one entry point less :-)

sp9bsl, 2018-02-15:
Hi Jason,
STM32F4 uses 32bit single precision FPU according to IEEE 754 with some extensions to double presision (64 bit). So, there is no benefit of using the 16 bit reduced precision (even if it would like to work at all) because all computations uses 32 bit float registers (sometimes paired to form a 64 bit register). The link you provided says that it is implemented in GCC but for NEON like core.

db4ple, 2018-02-15:
Don't bother to save 42 bytes. Only place where we are talking this level of reduction may the bootloader as we have a fixed size of 32k here and are pretty close to the limit. The main reason for the key beep to be dynamically calculated is the variable beep frequency. And we use the same sine table we use for generating side tone / CW IQ / Tune. So in terms of memory usage it is quite efficient as it is. 

And claiming something is wasteful needs proof that it indeed is wasteful. Did you measure the overhead compared to your new solution? And even this is kind of irrelevant, since if you save half the time, hey, that sound good but if this is 0.01 uS (10 ns) keep in mind that our audio interrupt uses between 33 uS and about 400 uS. So where would such an improvement get us? If the code gets less complicated for us mortals to understand, I am all in for any change in that direction even if there is a comparatively small (!!!) penalty in terms of absolute efficiency.

I think what is indeed relevant is to figure out is a good scheme for the loudness/volume regulation so that users can easily find an acceptable volume and the usage/visualization makes sense (the OFF vs. 0 thing).  I think a mapping between integer value used as index into a  float32_t array representing volume scale factors would the easiest way to tackle this and the amount of volume levels would not required us to build huge tables in flash memory. 

Background info:
Config values are internally stored as 16 bit values, not matter what. 
In the visualization of configuration values in some cases we replace the numerical 0 with "OFF". You'll find a number of places where we do that in the menu (see ui_menu.c).

db4ple, 2018-02-15:
BTW, for those playing along at home: audio_management.c void AudioManagement_LoadBeepFreq() reveals why zero is not silent. The original author (Clint or Chris) scales the 0 - 20 range into a multiplier (read gain) from 3/400 to 103/400. Here you can see why 0 cannot be silent. volume level 0 is just about 33x less gain than level 20, that is not much, our hearing is much better than that. The reason why there is a separate on/off for the keep beep is however obvious, you can set your preferred volume and then disable or enable key beep depending on your mood. BTW, I am not a key beeper at all.

DG9BFC, 2018-02-15:
i like the key beep ... reason is
i have set my sidetone for cw to 700 as i did with the key beep
now my ear is "trained" to that 700hz (i can tune in a cw signal by ear with a low error without tuning helper) ... finetune can be done with tune helper but when the ear is trained on the 700hz ... you automagically tune a cw signal to same tone (our ear easy can tune two tone to match with ease)
so ... that is only reason for keybeep here ... grin
sadly psk is fixed to 500 ... hmmm thinking of changing all to 500 here ... then only rtty would stick out of the frequency plan

frazierjason, 2018-02-16:
Thanks to both for painting a fuller picture.  I still stand by the idea that recalculating the float value on every key beep uses cycles that don't benefit the user, even if only consuming a few microseconds and associated power.  I had the same idea about using a small 21 element float table to reference directly from the stored integer setting, and the __fp16 suggestion was there merely to help save on flash usage.

Also, we could just recalculate only when we know the key beep frequency or volume was changed.  Whoever wrote that code has the same leanings because it's right there in AudioManagement_KeyBeep() as comments to consider reducing calls to LoadBeepFreq().  This seems tricky to do, and I haven't read all the audio mgmt code yet, so I would probably not try for this in the first go.

Also, I have been playing with alternate scaling methods using integer division in an Excel table to see if, say one divided by an integer, could reasonably approximate the very linear float table in the code now.  It's a little jaggy but could be smooth enough in all but the loudest few volume steps.  If it works, then the stored setting could be used directly to make the float at time of applying the volume multiplier in the audio code.  I'll try hacking in my alternate integer-derived table on my own fork, give it a listen among a few friends from speaker and a pair of big headphones, and see if the volume approximations sound linear "enough" during increase or decrease to be usable.  If they work out OK, I'll make some further proposals.

Regarding __fp16, it is a type used for data storage, and my intent was to save on flash usage.  ARM automatically converts them into float32_t (it must!) before they are used in any math operations.  But I wasn't aware that this feature is lacking in Cortex-M4, which has only the FPv4-SP unit.

sp9bsl, 2018-02-16:
To be honest, it would be best to have dB scaled factor there, referenced to current volume setting, it could be changed with 3dB steps or so. All calculations regarding the beep_loudness_factor should be moved to some init function and it should be recalculated everytime when changed the menu item. Creating a LUT is not the best idea in my opinion. Remark: we use dB calculations in few places (example: spectrum display) so, this function is already linked in code.

db4ple, 2018-02-16:
Hi All,
a) You're right, in the code is a comment wondering if we really have to call AudioManagement_LoadBeepFreq() inside the AudioManagement_KeyBeep() functions.  You're wrong, this was not the comment of the author of that function, it was my comment.  When I refactored the code, and reviewed large portions of the code, I found this to be an potential improvement but decided not to follow that lead due to not much improvement in the overall scheme. Just comment the call out and see if the beep still works and if beep frequency changes in the menu will be immediately applied. 
Here we could indeed save a few cycles. But nothing anyone will ever notices I suppose.

b) All what is left to do in the audio interrupt is to fetch an integer value from a table in flash using an index which was converted from a float, convert the fetched integer to a float value (using an conversion opcode of the MCU), multiply this with another float, add this result to another float, add another float to the accumulator).
To give you an impression of what times we are talking about, have a look here:
http://www.micromouseonline.com/2011/10/26/stm32f4-the-first-taste-of-speed/

A float mult is about 6 cycles,  a float add 3 so we are talking maybe about 20-30 cycles per sample and it runs 32 times (this is the blocksize) so lets assume this takes 1000 cycles, which translates to about 6uS. We do this 1500 times a second, so it is 9ms out of 1000ms, or roughly 0.9 percent. Cutting this down by half would reduce the load by 0.45 percent. Just to give you an idea what we are talking about. The whole audio interrupt probably does hundreds or thousands of float multiples/adds each time it runs in the digital filters. This is where time is "lost".

c) However, I don't understand how you come to such a short fixed length table for generating the variable frequency output. 

I won't deter you from anything you want to do, it is a free world. And it is a lot of fun to tinker around with code and make it perform better, I know that.  Happy to be proven wrong and see us gaining significant reductions in audio driver load with your changes. 
Show your code and your numbers and we will see.

frazierjason, 2018-02-17:
It's not time but watts that I am interested in recovering.  Usually there are only a few big easy wins in cutting power usage, and after that, its lots of little changes with all risk and no glory that add up to real power savings.

I understand nobody asked me to come in here and change code that was considered to already work just fine.  I want to get back into embedded C, and this project looks really fun, and I'm too new of a ham to have strong opinions or experience on most features users might be asking for.  Embedded handsets, RTOS profiling, and power management SDET work was part of my day job.  I think I could get into contributing on these aspects, and help hams either operate for longer, or shrink their battery, or both.

Based on comments here, small changes in the audio management code and related variables have some hidden consequences beyond their own scope.  I will probably set aside some of my ambitions, and work on other stuff until I've had time to more comprehensively absorb the audio design, all its members and methods, and how/when it uses them.  Per guidance here, I might at least try for the optimization of the "OFF" function, and maybe an optimization to apply the volume multiplier with less work.

I'm not proposing a big table to hold all of the possible variable frequency+volume combinations, just one to eliminate four math statements and two assignments just to calculate 0..20 into a float for the volume multiplier.  And yes I will check that fewer instructions (or less power cost for the instructions used!) are consumed in the new proposal.  Would anyone be adverse to reducing the steps to 19 values, plus OFF?  We would not lose audio volume range, but the steps would get less linear and a little louder at the very top 17..19 range.  You'll find the resulting floats of dividing each of these by one are within 98% of the original float value, except the last few.  If 21 steps are still required, I could add one more quieter value, plus a MAXINT-like value that tells us to be "OFF".  133,123,100,76,57,43,33,26,21,17,14,12,10,9,8,7,6,5,4.  Again, this is predicated on less cycles used to directly divide the stored volume integer setting by 1 to get our volume multiplier float, right there at the point of use in AudioDriver_RxProcessor(), without needing the calc variable or the ads.beep_loudness_factor global that get handled in AudioManagement_LoadBeepFreq().  No need to call any other dB calculation code either.  If closer linearity is desired throughout, I could scale up the integer values, and divide by four.

sp9bsl, 2018-02-17:
My proposal for use the dB scale is a thought regarding Sigi comment about his loud perception of beep when value is 0, because our "embedded receiver" called hearing has non linear scaling. Instedad of making weird calculations we should use let's say -60...-20 dB...
And regarding the "cycles": if code is properly optimized, TRM for M4 says: VADD=1 cycle, VMUL= 1 cycle, VMLA= 3 cycles, plus of course the small overhead for storing/reading from ram.

DG9BFC, 2018-02-17:
i do not need an exact db setting ... i do not need that is is an exact linear function (actually NOT linear ... finer steps on lower end and coarser steps on upper end would better match)
what i would need (or like to see after the work on keybeep) is
take the values that we have now from 0-5 (higher not needed) ... divide the "strenght by 4 or 5" ... and then scale that to 0-20
means setting 4 or 5 is what we have now at 1 ... with a few steps lower possible ... and when cranked up to 20 we have same level as we have now at 5 (that is loud enough to hear even in on a noisy fieldday)
so ... 4 or 5 times finer steps ... and only 20 or 25 % level of what we have now
maybe the lower end could be spreaded then a bit (not linear over the range but lower level and finer steps on lower end) ... how you solve that in soft ... i have no idea ... i just can tell how "loud" the beep is in level 0 or 1 and that could be setting 5 or 10 after the new scaling (so then low settings of 1 or 2 would be MUCH lower) .... and yes ... zero really should be zero (on off setting can be left separate)

df8oe, 2018-06-26:
There is a fix for "0 keeps key beep on" now (commit #6e52b9a6) in version 2.9.41.