df8oe/UHSDR #1072: Codec sample rate setting

marcotrevisan, 2017-09-15:
Hi all, 

   I'm considering building a mchf kit.
I see the audio codec has been set to 48 KHz. The WM8731 chip supports higher sample rates too, for example 96 KHz (at a first look at the datasheet custom values can be set as well, i.e. 100 KHz).

Is it feasible to parametrize (or at least change) the codec sampling rate? That would make it possible to increase the spectrum bandwidth (useful when you control the radio via USB for example), or adapt to slightly different codec chips... 
That may also require slight changes to the Tayloe detector capacitors to effectively use the increased sample rate. 

Apologizes if you already discussed this, but I get no results when I google on this topic for the mchf.

Cheers,
   Marco (waiting for ham license)

df8oe, 2017-09-15:
Hi Marco,

yes: WM8731 can operate at 96KHz. But I am not sure (better: I do NOT believe) that small MCU STM32F4 using only 192KB RAM is capable to do that. We are already at the edge of RAM and fighting for every usable KB.

But new starting OVI40 project using STM32F7 @ 512KB RAM of course will do that.

73 de
Andreas

marcotrevisan, 2017-09-15:
Hi Andreas,

  thanks for your reply. I'm eager to learn more about the OVI40 project ;-) 
By the way the STMF7 is reported to be pin to pin compatible with the F4... what if I try to plug a F7 instead of the normal F4? 
The price difference is not a budget breaker, it could be a good improvement if ther aren't some "showstoppers".

73!
Marco

df8oe, 2017-09-15:
There is no "drop in compatibility"....

73 de
Andreas

sp9fkp, 2017-09-19:
Hello !
I want add my wish to possible change samplerate also with change resolution to 24 bits. In this case (as option of course) can by switch demodulation off (for save processing) making  TRX as  specialised modem with CAT and USB interface to PC.

marcotrevisan, 2017-09-19:
Hi sp9fkp,

  I asked m0nka about this a while ago and he told me he once tried but got even more noise in, probably due to the increased i2s bus activity.

I don't know wether or not the design of that part is still like when he tried. Consider that 16 bits is quite a large word anyway, and the chip is internally doing oversampling. 
Anyway, doubling the sample rate while keeping the  same filtering passband leads to a 6dB increase of the snr and dynamic range if my memory is not fooling me, because the sampling white noise is spread in a wider spectrum ;)

df8oe, 2017-09-19:
Additionally if you are working < 15MHz the noise of antenna etc. is the edgde - not the sampling rate. Above 15MHZ I do agree: more bits == better sensitivity...

I2C-speed is always the same. Data is transferred via I2S. I2S does not interfer rf reception.

Again: I am not sure F4 will do this job and running as a standalone SDR at the same time...

DD4WH, 2017-09-20:
The only two advantages of a higher sample rate I see are:
1.) the extended spectrum/waterfall display width (48kHz with 48ksps sample rate, 96kHz with 96ksps)
2.) if we could overclock the codec and bring it to 192ksps, we could demodulate wide FM radio stations in undersampling mode (even in Stereo!) [you need about 180kHz bandwidth to demodulate wide FM Stereo]

However, that would afford a major refactoring of the software and it is totally unclear whether overclocking the WM8731 to 192ksps is possible. We did overclocking on the SGTL5000 and that worked quite well in the Teensy Convolution SDR for receiving wide FM Stereo radio with a very similar frontend hardware as the mcHF.

The major disadvantages are that:
- with higher sample rates you would have half or a quarter of the time to process your audio samples. So you would have to decimate your samples which would in the end lead to exactly the same results as the lower sample rates
- the antialias filters in the codec are designed for 48ksps, I think. So we should have some attenuation in the >48kHz range if we use 96 or 192ksps sample rate

Re 24bits: I think, it is useless to use more than 16bits in audio sampling ADCs. The effective number of bits is considerably lower even if we use 16 bits. You would never be able to build a hardware that is so noisefree in order to have an effective number of bits that exceeds 14. Even professional receivers in the 4000$-range do not use more than 17 or 18bit ADC converters (have a look at the absolutely outstanding receivers at http://www.reuter-elektronik.de/html/produkte.html). For example, the effective advantage of having a 16bit vs a 14bit ADC converter in the Reuter Pocket means an effective advantage of only 2dB in SNR and dynamic range . . . this is because of all the noise present in the analog and digital hardware, so the theoretically possible 6dB can never be achieved. And it gets even worse with an extension from 16 to 17 or 18bits . . . not to speak of 24 bits. I think with frequencies >15MHz the SNR is limited by the noise in the preamps . . . but I could be wrong here

Even if increasing the sample rate would lead to higher SNR (because we use shielded hardware components and blocks), I think that advantage would immediately be eliminated, because all the digital filters are designed for 60dB stopband attenuation ;-). The advantage of having such a "low" attenuation is the very steep rolloff of the filters. And no one has ever complained about the stopband attenuation being too low :-).

73 de Frank

K2AAE, 2017-10-13:
Take a look at CS5381 A/D: 216khz at 24 bit with 120dB - Matched with a Codec hub that has three digital inputs and seven DAC + PCM. WM8998 with the H7 at 400Mhz with 1Mb sram, interesting experiment. HMCAD5831LP9BE $2,905 26GS/s only 3 bits to screw with
24 bit octal days again.

df8oe, 2017-11-04:
We will test some day if 96KHz sample rate works on mcHF.

closed 2018-01-30 by df8oe