df8oe/UHSDR #1073: Keyer is clipping dits at higher speeds

ve3ids, 2017-09-15:
In the recent builds, the keyer is clipping dits at speeds over 20 WPM. I have went back to .57 and it works great there. Version 6 hardware. 

73 Don ve3ids

db4ple, 2017-09-15:
Hi,
you could help us a great deal if you would try to identify the last working release.
It should be easy and requires about 6 firmwares to be loaded if you go by a method called binary search:
We know 57 is not affected and 97 is. So you test (98 - 57)/2 + 57 = 41/2 + 57 = 77.5 -> 
If 77 does not have the problem it came in between 98 and 77 -> (98 - 77)/2 + 77 = 21/2 + 77 = 87.5 
if 87 does have the problem the answer is in between  87 and 77 -> (87 - 77)/2 + 77 = 82 
if 82 has it ....
You get the idea. So you will find quickly the right binary with the help of a little math.
Once that is done carefully, we have found the problematic commit and can start looking into the cause with much less work.

TU,
Danilo

ve3ids, 2017-09-15:
Hi Danilo,

Thanks for the interest. I have run through the commits and .84 seems to be
the problem. Up to and including .72 the keyer works perfectly. At .73, it
locks in tx mode, not actually transmitting but the display is frozen and I
have to cut the dc to turn it off. This happens through .83. At .84
forward, the keyer will clip the first dit in a letter at higher speeds. 72
is the last stable commit to not lock up and after 83, clip the start of
the transmission.
Hope this helps. Good luck!

72 Don VE3IDS

db4ple, 2017-09-15:
Thanks,
I hope this narrows it down for us (and you). 
@phaethon: Please consider this issue, seems to be related to the cw_keyer changes.

phaethon, 2017-09-16:
@ve3ids what keyer mode do you use? Does it start transmitting after a small delay (i.e. not transmitting the start thus clipping it) or it starts transmitting right away with a shorter dit?

s53dz, 2017-09-16:
To my view the clipping of **the first dit** is happening at all speeds, notable from 10 to 48 WPM.
As I can tell the issue is worst at a very-very short touch of a paddle. I can confirm it happens to the first one also when a train of dots is produced. 

The reason for this could also be in the anti-bouncing measures not being strong enough.

I have measured the signal at the TX mixer output. (D2.5.98)

73 Bojan

ve3ids, 2017-09-16:
I use iambic A mode. I am not sure whether it is the transmitter being
delayed in starting or not but that would be my guess but I am not an
expert by any measure. It sounds like if you had a dirty contact in a key
but only on the first dit. This is the same effect that the straight key or
external keyer has only with those it is much much worse.

72 Don ve3ids

phaethon, 2017-09-19:
@ve3ids , please, check the latest code

ve3ids, 2017-09-19:
Ericks

I tried .104 last night and it was improved but still randomly clips a few
first dits. I will try .105 tonight after I get home from work. Thanks for
your efforts!

73 Don ve3ids

s53dz, 2017-09-19:
D2.5.105,
My short report: All first dots in my case are clean now.
However, when testing the first dot I managed to "hang" mcHF on TX (PTT is holding but no RF output) with one really quick touch of the dot or dash paddle. But it is back to normal with the next CW element. For me this is ok.

TNX 73 Bojan

db4ple, 2017-09-19:
Nothing beats measurements: I measured the output of 2.5.103 and 2.5.105 at 40 WPM (Iambic, measure via second receiver).
2.5.103: First dit 26ms, second dit  31.25
2.5.103 First dit 31ms, second dit 31.25 
40 WPM  translate to exactly 30ms dit according to the formula, but we have slightly longer dit and a slightly shorter pause, so everything is back to normal.

I would say, we can close this issue now.

ve3ids, 2017-09-19:
I just tested the internal keyer. It is back, it works great again! Thanks
very much for your work and expertise, much appreciated. I am assuming this
is unrelated to the straight key/external keyer support? #1011?

Thanks again and 73
Don ve3ids

s53dz, 2017-09-20:
> Nothing beats measurements: I measured the output of 2.5.103 and 2.5.105 at 40 WPM (Iambic, measure via second receiver).
> 2.5.103: First dit 26ms, second dit 31.25
> 2.5.103 First dit 31ms, second dit [...]

This is of course very nice measurement and straight forward result, but when I tested it I realized that some individual first dots were differently cut resulting in their different length. And this issue is gone for me, which is great.

73 Bojan

closed 2017-09-22 by ve3ids