df8oe/UHSDR #1076: AGC Problem

g7jiq, 2017-09-15:
Not wanting to pick fault as its a great little radio with a good software team.

I have noticed a slight problem or querk with the WDSP AGC,

When in WDSP AGC mode,  and receiving AM. 
When a strong signal/noise/lightning or quick pulse of signal is recieved the audio drops about 50%. (even turning a light switch or mains socket does it) but doesn't return.

I can continue to hear stations and tune around, change bands and it stays at half volume.
SSB volume stays the same and is unaffected, when the band is changed back to the band receiving AM the volume drops again, without any noise pulse being pressent to trigger it, like it remembers its volume was low.
If I touch the AGC adjust up or down the volume returns to full level.
This happens at any WDSP AGC adjusted level.

Is this a setting I have missed or a bug ?

df8oe, 2017-09-16:
I have tried to reproduce but no luck. Audio is returning to level before strong splatter signal (or lightning pulse) in some seconds.

There must be something that differs between your and mine configuration (hardware / firmware version / settings).

73
Andreas

db4ple, 2017-09-16:
MIght be the case, that my AGC WDSP bugfix helped here as well.

g7jiq, 2017-10-04:
I have just re-tried the AGC WDSP and all seems well.

Thankyou

closed 2017-10-04 by g7jiq