df8oe/UHSDR #1079: Frequency display - explanation

s53dz, 2017-09-15:
Just for information and to avoid the confusion in operation (D2.5.98):

I am using a `CW display=SHIFT`
With this setting I get the proper frequency display for CW and SSB.
An example for CW:
f_rx_input_signal=28.100.000
frequency_display=28.100.000
f_tx_output_signal=28.100.000

**But the frequency is correctly shown only with `CW LSB/USB=AUTO`**

If you now change CW-L <--> CW-U, also the spectrum switches correctly.
And if you have calibrated the mcHF frequency (+-ppm) then also the signal sound stays the same.

73 Bojan

db4ple, 2017-09-15:
Could you please explain, to what issue is this comment related or what issue it describes?

s53dz, 2017-09-15:
Well, it is an information, as it says. And it solves a possible issue when the displayed frequency could be wrong!

It is connected to two issues:
1. issue of showing the frequency in numbers and in spectrum/waterfall
2. issue of using automatic choosing CW-U and CW-L. **Which you have solved but as I say if it is not switched back to AUTO, the spectrum and frequency display is wrong!**

73 Bojan

df8oe, 2017-09-16:
Sorry - I do not understand what is going on. My frequency digits and spectrum/waterfall frequency ruler does always match regardless CW-L/CW-U or any CW LSB/USB setting. Please explain what is not matching under what circumstances.

BTW:
I naver have had switched CW LSB/USB to anything other than AUTO since I do have mcHF (since ~2.5 years). The last minutes it was the first try and all is working as expected. But setting it to anything other than AUTO is senseless for me...

73
Andreas

s53dz, 2017-09-16:
Andreas,

I think you do not play enough with different settings of your mcHF, hi! Perhaps you miss something good when omitting something you think is not well. Like waterfall vs spectrum. I prefer spectrum!

But this story is a very simple one: 
I have always had the setting CW LSB/USB=AUTO, but when the tune helper worked only in CW-L, I switched CW LSB/USB from AUTO to LSB.
If you do so, please do not tell me you get a proper frequency display and a proper spectrum showing the same signal on the right position when switching between CW-L and CW-U mode.

On this occasion I also ask if we need the CW-L and CW-U at all? As a asked already when I reported the helper works only in CW-L. L and U modes have historical values only, in SDR of a less importance.

73 Bojan

df8oe, 2017-09-16:
Hi Bojan,

yes - it is **forbidden** to change any seting which satisfies me. My lifetime on earth is too short to waste time with something like that, hi...

Back to your answer:
##########################
I have always had the setting CW LSB/USB=AUTO, but when the tune helper worked only in CW-L, I switched CW LSB/USB from AUTO to LSB.
If you do so, please do not tell me you get a proper frequency display and a proper spectrum showing the same signal on the right position when switching between CW-L and CW-U mode.
##########################

But I tell you I get a proper frequency and waterfall showing!! If I am running in CW-L the signal is on the left side and when I switch to CW-U it is on the right side (offset is sidetone x 2). That is physically absoultely correct and reception, frequency digits and waterfall do match 100%.  (I like waterfall much more than spectrum - I never use that). I must retune to get the signal again - but that is normal. Please explain when you are using CW LSB/USB in setting LSB what does not match. I cannot find it :/

We need CW-L and CW-U to fastly change to CW if you receive a interesting CW signal in SSB. If you are tuning around the bands and you hear an interesting DX station you want to jump FAST from SSB to CW without loosing the signal. If I do not select CW LSB/USB=AUTO there is a chance of 50% that you loose the signal and must retune. So everything except AUTO ist absolutely senseless for me and I never use it.

73 de
Andreas

s53dz, 2017-09-16:
Andreas,

I must agree with you about the lifetime. Time is running to quick for all the ideas to come true.

Again, to reproduce the issue:
- CW display=SHIFT
- CW LSB/USB=LSB
- input signal=7.025.000

When in CW-L and tuned to input signal the frequency shows 7.025.000.
When in CW-U you have to re-tune, as you said, and then the frequency shown is -2*sidetone, i.e. 7.023.500 which is false. There exists no input signal.

But, as I said if CW LSB/USB=AUTO everything is ok.

BTW: 
If I am in a CW mood I always choose CW mode and then use different filter if needed. If I feel SSB then I choose SSB mode with the appropriate SSB filter. But this is of course anyone's personal preference.

It is like switching the LSB <--> USB. Some like staying on the modulation when switching, but you have to fine re-tune so and so to get to the right frequency.

73 Bojan

df8oe, 2017-09-16:
Again there is a a setting I never would use: "SHIFT".... I tried and could confirm an isue!!

If you select "CW LSB/USB Select LSB"
*AND* "CW Freq. Offset SHIFT"
and you operate in CW-L RX is working as expected. If you switch to CW-U now by pressing mode button long frequency display behaves as expected, reception marker line and frequency ruler, too but reception itself is now reduced at diedtone frequency.

You see: It is important to post EVERY relevant setting. The missing magical word was "SHIFT".

73
Andreas

s53dz, 2017-09-16:
Danilo,
FW D2.5.101 - the issue for me is gone.

Andreas,
I am attaching a table showing the CW_display settings results as with D2.5.101.
I wonder, how come, you "never would use" a SHIFT CW_display mode?

73 Bojan

[CW_display.pdf](https://github.com/df8oe/UHSDR/files/1308443/CW_display.pdf)

closed 2017-09-16 by s53dz

df8oe, 2017-09-16:
The reason is simple:

When I am tuning around the bands looking for an interesting station I *always* use SSB mode. In SSB mode filter bandwidth is big enough to fit any possible mode. If I catch a CW signal in the middle of SSB area I want to switch as fast as possible to CW. If I use setting "SHIFT" I loose signal when I switch to CW. I have to retune - very ugly. If I use setting "RX" signal stays and only bandwidth of filter is reduced. OK - frequency display changes in mode RX - but who does it care? Important is that I can toggle through the modes and don't lose signal!

df8oe, 2017-09-16:
I cannot confirm your attached pdf.

I must retune in SHIFT - not in RX.

db4ple, 2017-09-16:
Hi Andreas,
retune@S53DZ -> Retune when switching CW-L and CW-U
retune@DF8OE -> Retune when switching from SSB  to CW (same sideband)

You cannot have both at the same time (yet).
I was thinking if we would make CW Sideband switching in RX so that the other sideband is tuned to the same sidetone frequency as the one we are coming from. Right now we have an offset of 2xsidetone frequency.

df8oe, 2017-09-16:
:)

Thanks Danilo. I never have had switched CW-L or CW-U manually while working on the bands. It is senseless for me. But it is important not to do a retune when switching modes...

But that is "way of oerating". In my view SHIFT is a function without any use - in the view of others it is RX or TX. So *all* settings are neccessary.

s53dz, 2017-09-16:
Andreas,

I agree with you about seamless switching between CW and SSB mode, of course. And for this the RX mode is favorite. But it puzzles me what you said that one is happy with the CW_display=TX mode. This for CW means, as my table says, that you are receiving and transmitting CW on a different frequency from the one that is shown. And that the signal frequency shown on spectrum does not correspond to that shown in numbers. Whereas in RX or SHIFT mode you always have CW RX and TX on the displayed frequency. (Same is true also for SSB.)

And if somebody says to these hams, PSE QSY to ...? OK, they will meet somehow, since the frequency display is not important, is it? Hi.

Probably you should take this difference into account by proper displaying different TX and RX frequencies in the two frequency numbers.

So, at the end of the day we are all happy. And the mcHF remains a proactively excellent machine.

TNX 73 Bojan