df8oe/UHSDR #1086: Warning low(weak) tension

ON5ZA, 2017-09-19:
Hello Andreas,
I come by this circuit because I do not find the page to propose 
modifications.
He would not be interesting of indicate a voltage drop of the food to 
leave for example 11 volts or others by
  indicating him by one red blinking of "V" 
Thank you for the Gustave ON5ZA

s53dz, 2017-09-26:
For better wording I would propose this issue name to be changed to "Warning on low supply voltage". It could also be an extension of an existing use of switch-off at low voltage.

73 Bojan

phaethon, 2017-10-01:
Existing voltage indicator implementation already change color based on thresholds which are calculated from switch off voltage. You can disable low voltage switch off, and play with with the switch off threshold to see how it works. Isn't this what you want?

ON5ZA, 2017-10-01:
Thanks to all. I know that this change fe color of the tension exists. But I would wish that has to leave a certain low threshold "V" begins flashing in red. Simply that when you are qso there the change of color is not enough and your attention is not attracted. The blinking would be better a lot. ON5ZA

closed 2017-10-01 by ON5ZA

phaethon, 2017-10-01:
Actually, it is flashing (also, green LED is flashing, as LED could be off), when the low power shutdown is scheduled. So, it is more a design decision when to flash. Current logic is that if it is important, then shutdown is enabled and then it is flashing. If it is not so important, then it is more informative and just a change of colour is more appropriate.