df8oe/UHSDR #1087: RIT range extension from +-1000 to +-1200 Hz

s53dz, 2017-09-20:
Comment:
The mode UP1 is a very commonly used split mode of DX stations and means that the station operator expects calls at +1kHz from his TX frequency. If we add about half of an average CW BW filter range we end-up with +-1.2 kHz usable range.

In the lack of XIT function we could use -1kHz RIT in the above example.

73 Bojan

df8oe, 2017-09-20:
Yes, I know. I am using SPLIT in this case. Much more comfortable and flexible...

73
Andreas

s53dz, 2017-09-20:
Yes, SPLIT is universal and a very useful function.
But, with RIT/XIT you have:
- a constant inherent high resolution,
- an instant access to it.

73 Bojan

s53dz, 2017-09-30:
I opened this issue also to try to initiate an idea of having XIT as an alternative function on the same encoder. However it seems to be no interest for this.

I am closing this issue since the RIT range extension from the headline is easy to do in the file:
drivers/ui/ui_configuration.h
by changing
```
#define MIN_RIT_VALUE       -50     // Minimum RIT Value
#define MAX_RIT_VALUE       50      // Maximum RIT Value
```
to
```
#define MIN_RIT_VALUE       -60     // Minimum RIT Value
#define MAX_RIT_VALUE       60      // Maximum RIT Value
```
By trying this change I could not find no troubles.

73 Bojan

closed 2017-09-30 by s53dz