df8oe/UHSDR #1090: CW decoder issues

I2KBD, 2017-09-28:
After several tests with CW decoder, the decoding rate is very poor. The red LED is flashing in absolute synchronism with CW signal and the speed (after some time) is closely locked to actual speed. But the decoded text is nearly garbage. I connected the DFU USB port to my PC running MRP40 morse decoder software that copy perfectly after just a couple of seconds of speed setup. Is the work on this issue still underway? I think that good CW copy (like Elecraft KX3) would be highly welcome...!

Alberto I2KBD

DD4WH, 2017-09-29:
Hi Alberto,
you are probably right with your observation. I have been working on implementing an AGC for the CW decoder the last two weeks and was experiencing severe problems with decoding with the recent version. [my changes have not been submitted yet, so they are not appearing in the daily builds]
Probably there has been something broken in the last commit about 15/16 days ago.
Would it be possible for you to test which version works better than the recent one?
It probably happened between Sept 10th and 12th . . . 
73 Frank

I2KBD, 2017-09-29:
Hi Frank,
yes the first release of CW decoder were working far better. I will check when the problems started to show-up...
73 Alberto

df8oe, 2017-09-29:
Hi Alberto,

we must pick the best of both stages:
- the first releases has decoded much better - but they were unstable and sometimes firmware does hang
- the last releases do not hang under any circumstances but they do not decode very well

Lets fit the fortunes together ;)

73
Andreas

I2KBD, 2017-09-29:
Hi Andreas,
completely agree with you!
My concern is that the LED is flashing following CW signal 100% without spikes and speed tracking seems to be working (even if sometimes jumps to 70-80 wpm) so it seems to be a code interpretation issue...
I see some slow down on firmware update, are you busy on OVI40 development (I am just looking for board availability!)?

73 Alberto

df8oe, 2017-09-29:
Hi Alberto,

I am busy with OVI40 related things - and I am happy that I must not release every few hours new bleeding edge builds of firmware :) Of course we are working on firmware continuously and the next bleeding edges will appear sonn ;)

73
Andreas

DD4WH, 2017-09-29:
Hi Andreas, hi Alberto,
we were at a stage with the CW decoder (about two weeks ago), where:
- CW decoding worked very nicely
- hangs (where the hardware refused to work) were eliminated by Danilos´ great work
- but still sometimes the decoder went to very slow or very fast WPMs, when the threshold was not properly adjusted (too low, so that many non-CW events triggered the decoder)

So, we would have to begin again at that point of the development and implement an automatic threshold correction that eliminates false non-CW trigger events and implement a reset function for the CW decoder (just in case).
 
We just have to find that point and return to that stage.

73 Frank

DG9BFC, 2017-10-01:
hmmm ... an auto threshold would be fine but that also should be "tune-able" by the user
i found out that the fixed threshold works fine if the signals are more or less stable ... but if you have a bit of qsb you always have to be inside menue to turn the value up or down (and lower signals are badly decoded anyway) ... it even works together with the cw peak filter or together with noise reduction ...
(by the way ... could we also implement nr + peak like we have nr + notch????)

df8oe, 2017-10-01:
I have played around a little bit. For me 2.5.89 was the last well decoding version. Starting with 2.5.90 decoding was *much* worse...

Can you confirm?

73
Andreas

DD4WH, 2017-10-01:
Hi Andreas,
that is puzzling, where can I find the version numbers in github? Can you state the date and commit number? 
When I look into the commits I found that between daily 89 (11. Sep 8:06) and daily 90 (11. Sep 12:13) the only change was your change on touchscreen, which has nothing to do with CW decoder, so we are probably talking about different commits/versions here . . . 
73 Frank

DD4WH, 2017-10-01:
current version decodes quite well with manual threshold adjustment . . . BUT: the speed of WPM adjustment is TOO FAST, so it occasionally drifts away even when you have a very good signal.
We have to work on this

I2KBD, 2017-10-01:
Long test with decoder version 2.5.111
CW decoding is MUCH better than before. 
Had to lover the treshold to 10000 and keep AGC for decoder OFF. The issue seems to be the right speed identification. With a signal at 22 WpM, the software jumps between 15 and 29 Wpm. When finally stick to correct speed, the decoding is quite good (I still use MRP40 software connected to PC trough USB port that decodes correct 100% as reference ).

DG9BFC, 2017-10-03:
a small bugin menue for cw decode: if you select spike or short in spike cancel and then turn back to "off" the leading "S" is still shown on screen ... means you have to print "_______off" instead of "__off" on screen

DG9BFC, 2017-10-03:
another one: cw decoder is still active in sam mode?!? (and sometimes produces carbage letters on screen)
ok on the tune helper in am and sam .... shows you the carrier point very accurately

DD4WH, 2017-10-03:
- small bugfix "S" in "SPIKE" --> done
- CW decoder can be enabled in SAM and AM! Use menu entry in CW menu to enable/disable
- you will soon be able to switch off the LED with a menu entry

DG9BFC, 2017-10-03:
after a short test ... cw decoder works fine (most of the time) ... sometimes decoder agc on is better ... sometimes better use manual threshold

I2KBD, 2017-10-04:
Yes, I noticed the same. Decoding stops and then print a full line of characters. I think it is important to be able to lock the  speed once the decoding is correct (a simple press of one of F1-F5 button) otherwise the software continues to adjust and often stick to wrong speed. Also manual speed selection is useful (you can identify the speed by listening most of the time) and allow a limited automatic adjustment (20%) around manually selected speed.

73 Alberto

S52CQ, 2017-10-21:
Hi, is it normal that set parameters are not saved? Namely I do set parameters for CW decoder, save them with a long press, but when I turn Off & On again, parameters are not saved.

db4ple, 2017-10-21:
Yes, normal. CW Decoder is still considered to be experimental, and we are figuring out which parameters we truly need and then make them stored. Due to the way we store parameters it is rather difficult to remove a parameter once it has been saved once. So we have to be sure it is needed as stored parameter.

df8oe, 2017-11-01:
Please close issue - CW decoding works well at the moment. All other goodies like saving setting will follow next time.

closed 2017-11-02 by I2KBD