df8oe/UHSDR #1093: ExtIO.dll for HDSDR

WY5Y, 2017-10-01:
Does anyone know which ExtIO.dll that the mcHF emulates? I've tried multiple .dll files, but no success yet. Any help would be greatly appreciated.

DG9BFC, 2017-10-01:
no need for an extio driver
hdsdr works with omnirig driver ... that connects to mchf with selecting ft817 as the rig
in hdsdr set "sdr is on if output that is controlled by omnirig"
set "sync in both directions"
set 12000 in if frequency box and global offset to -12000 (i prefer low side mixing to see upwards from my own qrg)
set 700 hz in cw box (all other offsets set to zero)
select mchf soundcard as input for hdsdr and set mchf to iq in out (DIQ)
thats it :-)
now the hdsdr screen looks like the mchf screen (you see sam spectrum and both stay in sync)
regardless if you turn vfo in mchf ... or change vfo qrg in hdsdr ... both are synced via omnirig driver
have fun ... sigi

WY5Y, 2017-10-01:
Thank you so much. I was trying to connect to HDSDR using Ham Radio Deluxe at the same time. I could change frequencies and bands, but waterfall and spectrum scope wouldn't work properly. Thanks again.

DG9BFC, 2017-10-01:
you are welcome :-)
if all is set properly then it sould work (it does here with hdsdr and multipsk)

DG9BFC, 2017-10-02:
it is also explained in the wiki
https://github.com/df8oe/UHSDR/wiki/USB-CAT-and-AUDIO-Mode

scroll down to centre of page "sdr programs to use with i/q data"

there it says hdsdr ... "rig control via omnirig"

you can use any cat software to control the rig (be it via omnirig driver or or via hamlib driver)

only trouble you will get is if two different cat software wanna talk to the same comport (usb connector represents a virtual serial port in windoze)

that is the reason why you should stick with ONE ham suite and not mix two of them

means ... an good one is like dx lab suite ... spotcollector and logging ... both work fine with mchf (and my favourite digimode sofzt multipsk also talks to that same logboog dx keeper in the dx lab suite)

i never used ham radio deluxe so can not help you with the setup

DG9BFC, 2017-10-03:
in my view ... issue can be closed

closed 2017-10-06 by WY5Y

undiplomatic, 2022-01-01:
I've been trying to configure mcHF to work with HDSDR but there is an issue with the offset. To get HDSDR to show the same spectrum the _RX TX Freq Xlate_ must be set to OFF in mcHF, which is not a recommended setting.

I followed Sieg's instructions and set the IF-Frequency in HDSDR to 12000 and the Global Offset to -12000 but now the tuned frequency in HDSDR does not match mcHF. I've spent quite a bit of time trying to configure this :/

db4ple, 2022-01-01:
Hi,
what is your setting of CAT-DIQ-FREQ-XLAT ? See https://github.com/df8oe/UHSDR/wiki/USB-CAT-and-AUDIO-Mode#please-read-understanding-and-configuring-external-frequency-control-via-cat and https://github.com/df8oe/UHSDR/wiki/USB-CAT-and-AUDIO-Mode#sdr-programs-for-use-with-iq-data for a discussion on this topic.

DG9BFC, 2022-01-01:
If you set offset in hdsdr abd mchf correct then they do match..If you have to set to minus or plus in hdsdr depends on how you set it in mchfI prefer to have the local above the rx cause then u see upwards from my own signal.. And dx stations usually work upwards split.Bo need for an extio driverOmnirig driver set to control the mchf via cat. Mchf soundcard select as input fir hdsdr... And set mchf to DIQ (digital raw iq data NOT AUDIO!)Try it.. Mchf and hdsdr can be well in sync... Clicking with mouse on screen in hdsdr or with finger on lcd of the rigIr tune vfo... In all cases all stays in syncDg9bfc sigi

DG9BFC, 2022-01-01:
Sorry for typos.. Typing on smartphone and pressed send a bit earlyAm 01.01.2022 15:23 schrieb ***@***.***:If you set offset in hdsdr abd mchf correct then they do match..If you have to set to minus or plus in hdsdr depends on how you set it in mchfI prefer to have the local above the rx cause then u see upwards from my own signal.. And dx stations usually work upwards split.Bo need for an extio driverOmnirig driver set to control the mchf via cat. Mchf soundcard select as input fir hdsdr... And set mchf to DIQ (digital raw iq data NOT AUDIO!)Try it.. Mchf and hdsdr can be well in sync... Clicking with mouse on screen in hdsdr or with finger on lcd of the rigIr tune vfo... In all cases all stays in syncDg9bfc sigi

undiplomatic, 2022-01-01:
It turned out to be a config issue in HDSDR, I had to reset all the HDSDR settings, start again, and it works.
It did take me several hours to get everything working, so to save anyone else the time, I put together a quick video, it's on YouTube now so should appear in any searches: https://youtu.be/cuMLVm1Uy3w