df8oe/UHSDR #1098: CW - red TX LED does not work right

s53dz, 2017-10-02:
D2.5.111:
Issue is there probably already from the introduction of CW decoder, which uses the red LED.

When in the CW mode the red LED on receive is showing the CW code. When switching to TX the red LED remains in the state as it was during the last received CW symbol for the period of transmission. It means lit or dark.

So, it can happen that at the transmission time there is a red TX LED remaining dark.

73 Bojan

DG9BFC, 2017-10-03:
an addition: if you push tune button it lights up for a short moment

DG9BFC, 2017-10-03:
correction ... now using 112 version ... and tx led is working when pushing tune button

df8oe, 2017-10-03:
I never never never have tried to TX in SAM - nice idea :)

s53dz, 2017-10-03:
Frank,

D2.5.116: 
Now the red LED works ok when the TuneHelper is on and in CW mode going from RX to TX. I like this to be sure that the TX is on.
But, I have a question, or I did not get it right. The red LED now is dark on RX. Which does not bother me. But is it the TuneHelper now on AGC?

And when the CW decoder display is on, red LED still does not work on TX.

73 Bojan

DG9BFC, 2017-10-04:
tune button in sam now is disabled :-( ... in all other modes tune shows red led (txing)

df8oe, 2017-10-04:
Why "sad smily"? What sense does TXing in SAM (please see ma post above)??

s53dz, 2017-10-04:
D2.5.118,
The issue as in the headline is fixed.

TNX Bojan

closed 2017-10-04 by s53dz

DG9BFC, 2017-10-05:
andreas ... that was a joke ;-) ... sure i will not try to do a qso in sync am ... but what if a user pushes that button (by accident)?!? ... now solved